optimsoc/optimsoc #14: verilator simlib --meminit does not accept absolut paths

anapetrovska, 2016-04-14:
When running 
~~~
tb_compute_tile --standalone --meminit=hello.vmem
~~~
The program finishes, but when running
~~~
tb_compute_tile --standalone --meminit=/some/absolute/path/hello.vmem
~~~
we only get
~~~
[              257314, 0] Software reset
[              257330, 0] Software reset
[              257346, 0] Software reset
[              257362, 0] Software reset
[              257378, 0] Software reset
[              257394, 0] Software reset
[              257410, 0] Software reset
~~~

and the program never finishes. Any ideas?

wallento, 2016-04-14:
Hi Ana,

yes, I think I know the source of the problem.The length is limited currently. I can fix this in a few minutes. Can you meanwhile verify that you are running verilator after version 3.882 or newer?

Thanks,
Stefan

wallento, 2016-04-14:
To be more precise, the problem is most probably here: https://github.com/optimsoc/sources/blob/master/src/host/sim/src/VerilatedControl.cpp#L74

imphil, 2016-04-14:
more than 16 characters seems to work, only absolute paths don't (or a different length > 16).

Verilator: same for 880 and 882

wallento, 2016-04-14:
The code needs to be changed for the new verilator feature (I added strings as parameters instead of files). I think its 16*4. How long is your path?

imphil, 2016-04-14:
We solved it for now using a relative path, so no hurry here if it needs to change anyways.

wallento, 2016-04-14:
okay, I will update it anyways the next minutes (packaging verilator currently) and lets see if it solves the issue.

commit fbc16de2 (2016-10-02): Adjust to simutil changes

We now can use arbitrarily-long strings as filenames for --meminit
passed to Verilator, due to using the correct data type.

Requires Verilator 3.882+ (no new requirement)

Fixes #14
closed 2016-10-02 by imphil