optimsoc/optimsoc #24: networkmanager_ct includes optimsoc_def.vh

imphil, 2016-09-30:
```
optimsoc-fusesoc --verbose --cores-root $HOME/OptimSoc/optimsoc-sources/examples/sim/compute_tile sim optimsoc:examples:compute_tile_sim --vcd --USE_DEBUG 1
```
[image]

wallento, 2016-10-02:
Are you sure you are on the correct branch etc.? `networkadapter_ct.v` is not supposed to even exist anymore. It is `networkadapter_ct.sv` now.

imphil, 2016-10-02:
It wasn't found by me, but by Kai. In my testing on master it works, so let's leave this open until I know what exactly goes wrong on his setup.

wallento, 2016-10-02:
It seems to me that the new examples were taken from the source distribution (see `cores-root`), but the old distribution files, i.e., `make` was not executed.

imphil, 2016-10-05:
Kai: did you run `make` after updating to optimsoc master?

Acetofive, 2016-10-05:
Hi,

I updated the repository and executed: -> "make clean" and then make again. The error still comes during the compilation. It looks like a nested include for me.

Acetofive, 2016-10-05:
Update: I tried it again in a new folder and the error message is still the same. I cloned:

`git clone https://github.com/optimsoc/sources.git optimsoc-sources`

set` BUILD_EXAMPLES_FPGA := no`

and executed a `make`

imphil, 2016-10-05:
You don't need to modify the Makefile, instead type
`make BUILD_EXAMPLES_FPGA=no`
But that's not the problem, please send me the full output of 
`make BUILD_EXAMPLES_FPGA=no >logfile.txt 2>&1`

Acetofive, 2016-10-05:
Hallo Philipp,

hier sind die logfiles vom Build und der Fehlermeldung.

mfg Kai

[1;37mINFO:  Preparing openrisc:mor1kx:cappuccino[0m
[1;37mINFO:  Preparing opensocdebug:interfaces:mor1kx_trace_exec[0m
[1;37mINFO:  Preparing optimsoc:core:mor1kx_module[0m
[1;37mINFO:  Preparing optimsoc:lisnoc:blocks[0m
[1;37mINFO:  Preparing optimsoc:lisnoc:dma[0m
[1;37mINFO:  Preparing optimsoc:lisnoc:mp_simple[0m
[1;37mINFO:  Preparing optimsoc:lisnoc:router[0m
[1;37mINFO:  Preparing optimsoc:networkadapter:networkadapter_ct[0m
[1;37mINFO:  Preparing optimsoc:bootrom:bootrom[0m
[1;37mINFO:  Preparing optimsoc:base:functions[0m
[1;37mINFO:  Preparing optimsoc:sram:plain[0m
[1;37mINFO:  Preparing optimsoc:debug_system:mam_wb_adapter[0m
[1;37mINFO:  Preparing optimsoc:wb_interconnect:bus[0m
[1;37mINFO:  Preparing opensocdebug:interfaces:dii_channel[0m
[1;37mINFO:  Preparing opensocdebug:modules:mam[0m
[1;37mINFO:  Preparing opensocdebug:modules:mam_wb[0m
[1;37mINFO:  Preparing opensocdebug:blocks:timestamp[0m
[1;37mINFO:  Preparing opensocdebug:blocks:tracesample[0m
[1;37mINFO:  Preparing opensocdebug:blocks:tracepacket[0m
[1;37mINFO:  Preparing opensocdebug:modules:stm_mor1kx[0m
[1;37mINFO:  Preparing opensocdebug:modules:ctm_mor1kx[0m
[1;37mINFO:  Preparing optimsoc:base:config[0m
[1;37mINFO:  Preparing optimsoc:tile:compute_tile_dm[0m
[1;37mINFO:  Preparing optimsoc:trace_monitor:trace_monitor[0m
[1;37mINFO:  Preparing glip:common:channel[0m
[1;37mINFO:  Preparing glip:backend:tcp[0m
[1;37mINFO:  Preparing opensocdebug:blocks:buffer[0m
[1;37mINFO:  Preparing opensocdebug:interconnect:debug_ring[0m
[1;37mINFO:  Preparing opensocdebug:blocks:regaccess[0m
[1;37mINFO:  Preparing opensocdebug:modules:scm[0m
[1;37mINFO:  Preparing opensocdebug:modules:him[0m
[1;37mINFO:  Preparing optimsoc:debug:debug_interface[0m
[1;37mINFO:  Preparing wallento:simutil:verilator[0m
[1;37mINFO:  Preparing optimsoc:examples:compute_tile_sim[0m

[1;37mINFO:  Verilating source[0m

[1;37mINFO:  Starting Verilator:[0m
[1;37mINFO:    Verilator working dir: /nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/sim-verilator[0m
[1;37mINFO:    Verilator command: /nfs/lis/tools/veripool/verilator/3.886/bin/verilator --cc -f input.vc --top-module tb_compute_tile --exe -LDFLAGS " -Wl,--start-group /nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/sim-verilator/wallento_simutil_verilator.a -Wl,--end-group " -CFLAGS -I/nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/src/wallento_simutil_verilator/inc -CFLAGS -I/nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/sim-verilator -CFLAGS -I/nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/src /nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/src/optimsoc_examples_compute_tile_sim/tb_compute_tile.cpp -GUSE_DEBUG=1 --trace -Wno-fatal -CFLAGS "-std=c++11" -LDFLAGS "-pthread"[0m

[1;37mINFO:  /nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/sim-verilator[0m
[1;37mINFO:      /nfs/lis/tools/veripool/verilator/3.886/bin/verilator --cc -f input.vc --top-module tb_compute_tile --exe -LDFLAGS " -Wl,--start-group /nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/sim-verilator/wallento_simutil_verilator.a -Wl,--end-group " -CFLAGS -I/nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/src/wallento_simutil_verilator/inc -CFLAGS -I/nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/sim-verilator -CFLAGS -I/nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/src /nas/ei/home/ga73sip/test/optimsoc-sources/build/optimsoc_examples_compute_tile_sim/src/optimsoc_examples_compute_tile_sim/tb_compute_tile.cpp -GUSE_DEBUG=1 --trace -Wno-fatal -CFLAGS "-std=c++11" -LDFLAGS "-pthread"[0m
[1;31mERROR: %Error: ../src/optimsoc_networkadapter_networkadapter_ct/verilog/networkadapter_ct.v:33: Cannot find include file: optimsoc_def.vh
%Error: ../src/optimsoc_networkadapter_networkadapter_ct/verilog/networkadapter_ct.v:33: Looked in:
%Error: ../src/optimsoc_networkadapter_networkadapter_ct/verilog/networkadapter_ct.v:33:       ../src/optimsoc_lisnoc_dma/rtl/dma/optimsoc_def.vh
%Error: ../src/optimsoc_networkadapter_networkadapter_ct/verilog/networkadapter_ct.v:33:       ../src/optimsoc_lisnoc_dma/rtl/dma/optimsoc_def.vh.v
%Error: ../src/optimsoc_networkadapter_networkadapter_ct/verilog/networkadapter_ct.v:33:       ../src/optimsoc_lisnoc_dma/rtl/dma/optimsoc_def.vh.sv
%Error: ../src/optimsoc_networkadapter_networkadapter_ct/verilog/networkadapter_ct.v:33:       ../src/openrisc_mor1kx_cappuccino/verilog/optimsoc_def.vh
%Error: ../src/optimsoc_networkadapter_networkadapter_ct/verilog/networkadapter_ct.v:33:       ../src/openrisc_mor1kx_cappuccino/verilog/optimsoc_def.vh.v
%Error: ../src/optimsoc_networkadapter_networkadapter_ct/verilog/networkadapter_ct.v:33:       ../src/openrisc_mor1kx_cappuccino/verilog/optimsoc_def.vh.sv
%Error: ../src/optimsoc_networkadapter_networkadapter_ct/verilog/networkadapter_ct.v:33:       ../src/optimsoc_lisnoc_dma/rtl/optimsoc_def.vh
[... cut ...]
(I)   + Copy build artifacts
(I)  + compute_tile_sim_quadcore (compute_tile_sim --NUM_CORES 4)
(I)   + Build
(I)   + Copy build artifacts
(I)  + system_2x2_cccc_sim_dualcore (system_2x2_cccc_sim --NUM_CORES 2)
(I)   + Build
(I)   + Copy build artifacts
(I)  + system_2x2_cccc_sim_dualcore_debug (system_2x2_cccc_sim --NUM_CORES 2 --USE_DEBUG 1)
(I)   + Build
(I)   + Copy build artifacts
(I) Build finished.
(I) All build artifacts are available at /nas/ei/home/ga73sip/test/optimsoc-sources/objdir/dist

imphil, 2016-10-05:
Was ist das oben für ein log von verilator/fusesoc?

Unten
```
compute_tile_sim_dualcore (compute_tile_sim --NUM_CORES 2) (I) + Build (I) + Copy build artifacts
```
sieht man genau, dass der Build des compute tiles aus den Quellen funktioniert.

Und wo kommt die networkadapter_ct.v Datei her? Die ist in den Quellen nicht vorhanden.

Bitte überprüfe auch ob deine FUSESOC_CORES Variable richtig gesetzt ist (d.h. keine eigenen Änderungen, nur das was optimsoc-environment.sh setzt).

Acetofive, 2016-10-05:
Ok, danke. Der Fehler lag daran, dass ich das optimsoc-environment.sh 
nach einem neuen "make" nicht ausgeführt habe.

Danke fürs Helfen.

Acetofive, 2016-10-05:
I didn't run the `optimsoc-environment.sh` to set the variables new after the update/make.

closed 2016-10-05 by imphil