optimsoc/optimsoc #30: Fix hello_mpsimple on 2x2 CCCC in Verilator simulation with debug system

imphil, 2016-11-22:
See https://api.travis-ci.org/jobs/177929879/log.txt?deansi=true

The same software works when loading the memories through --meminit and not through the debug system/loadelf.

imphil, 2016-11-24:
Actually it's independent of the application, `hello` shows the same bug: no code runs on tile 3 when initialized through the debug system.

wallento, 2016-11-24:
Thanks, this narrows it down. I will have a look after lunch

imphil, 2016-11-24:
Actually MAM is still writing to the memory of the last tile when we start the CPUs. The correct way to resolve this is to make sure that the memory write is finished before we do anything else. The incorrect way is to add a sleep in runelf.py.

wallento, 2016-11-24:
Oh, yeah, I actually had the solution in my mind for a while, which is acknowledging writes. I mainly thought about it for the synchronous experience on the host side, but now we have a very good reason to fix it asap. The sleep solves it for your measurements at least..

imphil, 2016-11-24:
Opened https://github.com/opensocdebug/hardware/issues/30 on the OSD side to get this fixed.

commit ad3742e7 (2016-11-29): OSD hardware: pull in synchronous MAM changes

This should be reverted back to the master branch once this branch is
merged with the master.

Fixes #30
closed 2016-11-29 by imphil