optimsoc/optimsoc #36: Build a platform with a host tile

wallento, 2017-03-08:
The idea is that we have one compute tile that runs Linux and has a UART interface or similar to the outside world.

wallento, 2017-03-22:
- [ ] Add UART (OSD dem_uart) to compute_tile and make it configurable (turns CT into HT)
- [ ] Demonstrate with system_2x2_hccc in simulation
- [ ] Demonstrate with system_2x2_hccc on Nexys4

wallento, 2017-06-20:
Please find that in branch `host-tile`

imphil, 2017-08-31:
Prototype code in host-tile branch, configuration needs rework to be integrated into the CONFIG structure and more generalized.

imphil, 2018-12-18:
We now have a single (multi-core) compute tile with Linux support. Having a setup with one "large" host tile and a couple compute nodes isn't there yet and mostly blocked by finding a good way to configure such a system.