optimsoc/optimsoc #42: Clock domain crossing at NoC interface of tiles

wallento, 2017-03-22:
Add a clock domain crossing at the NoC interface.

koenenwmn, 2019-02-09:
I was gonna have a look at this soon. A few questions though:

In the networkadapter_ct there are two signals that depend on the defines OPTIMSOC_CLOCKDOMAINS and OPTIMSOC_CDC_DYNAMIC. I couldn't find these definitions being used anywhere else, are they?
Also, depending on these defines the adapter has two output ports: cdc_conf and cdc_enable. What was the originally thought of purpose of these signals?

In general: If I understand it correctly we want to have the possibility to have compute tiles with different clocks from the NoC and from each other. Am I right in assuming the clock frequency of the tiles would typically be higher than the one from the NoC?

The easiest approach I can think of is replacing the output and input buffers of the networkadapter with CDC FIFOs and providing the module with two clocks. Or am I missing anything crucial that speaks against this approach?

imphil, 2019-02-11:
`OPTIMSOC_CLOCKDOMAINS` and `OPTIMSOC_CDC_DYNAMIC` are remains of two almost completely removed features.

`OPTIMSOC_CLOCKDOMAINS` was a define. If enabled, a CDC FIFO between the compute tile and the router was instantiated.

`OPTIMSOC_CDC_DYNAMIC` enables runtime (hence "dynamic") frequency setting of the core. Two CPU-writable registers, `cdc_conf`, and `cdc_enable` were provided by the network adapter which could be used by software running on the core to adjust the frequency the core is running at.

These features were never fully integrated and I don't think we ever had dynamic frequency scaling working at all. 

> Am I right in assuming the clock frequency of the tiles would typically be higher than the one from the NoC?

The answer strongly depends on the NoC type, in a unbuffered NoC you can easily have tiles with lower frequency than the NoC.

koenenwmn, 2019-02-12:
Ok, so I guess I can get rid of them then.

> The answer strongly depends on the NoC type, in a unbuffered NoC you can easily have tiles with lower frequency than the NoC.

You are right of course, but I was asking for the current NoC, since the implementation of the network adapter would have to be changed for another NoC type anyways.  
The question was more targeted towards on which side of the CDC FIFOs it makes sense to have the larger buffers in order to prevent starvation or backpressure.