optimsoc/optimsoc #70: FPGA examples don't work with the new NoC

koenenwmn, 2017-06-21:
The FPGA examples don't work with the new NoC yet. I tried to fix the issue in the 'noc-examples-fix' branch.
During synthesis of the system_2x2_ccc_nexys4ddr example I also got an error because of an invalid assignment to a non-register in 'noc_vchannel_mux.sv' lines 27, 28, 31, and 32.
After adding registers the synthesis succeeds. However, I now get an error during the implementation because the design requires more Slice LUT cells than are available in the target device.

@wallento is that something you could look into?

wallento, 2017-06-21:
Yes, i can look into it. Can you give me the error during implementation?

koenenwmn, 2017-06-21:
Thanks.
This is the error from the log file:

Phase 1.1 IO Placement/ Clock Placement/ Build Placer Device
ERROR: [Place 30-640] Place Check : This design requires more Slice LUTs cells than are available in the target device. This design requires 66443 of such cell types but only 63400 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.
ERROR: [Place 30-640] Place Check : This design requires more LUT as Logic cells than are available in the target device. This design requires 64642 of such cell types but only 63400 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device. Please set tcl parameter "drc.disableLUTOverUtilError" to 1 to change this error to warning.

wallento, 2017-06-21:
Thanks, looking into it now

wallento, 2017-06-22:
Reduced the LUT usage significantly with improved buffers

PR #73 (2017-06-22): NoC FPGA fixes
Fixes #70
closed 2017-06-22 by wallento