optimsoc/optimsoc #75: Random Crash when Tracing Execution on FPGA

ppenna, 2017-07-24:
**Description**

When deploying Linux and tracing the execution on the FPGA, random crashes occur.

This problem happens in the following situations:
- CTM tracing (no output is produced)
- STM tracing (no output is produced)
- CTM + STM tracing (expected output is produced)

**Reproducing the Problem**

1. Switch to the host-tile branch
2. Build a single-core configuration with a host-tile
3. Program the target device
4. Connect to the FPGA
4.1. opensocdebugd uart device=/dev/ttyUSB1 speed=12000000
5. Build Linux
5.1. git clone https://github.com/cart-pucminas/bluedragon
5.2. make
6. Load Linux into the FPGA memory
6.1. osd-cli
6.2. reset -halt
6.3. mem loadelf vmlinux 2
6.4. stm log stm000.log 3
6.5. ctm log ctm000.log 4 linux/vmlinux
6.6. terminal 5
6.7. start
6.8. wait for crash

imphil, 2017-07-24:
What's crashing? opensocdebugd? Did you try running under gdb and collecting a backtrace?

ppenna, 2017-07-24:
@imphil opensocdebugd is crashing when writing to the buffer.

I'll collect the traces with gsb and post them here.

ppenna, 2017-07-24:
Here is the backtrace when tracing CTM and STM

```
Open SoC Debug Daemon
Backend: uart
System ID: 0001
6 debug modules found:
 [0]: HOST
   version: 0000
 [1]: SCM
   version: 0000
 [2]: MAM
   version: 0000
   data width: 32, address width: 32
   number of regions: 1
     [0] base address: 0x0000000000000000, memory size: 134217728 Bytes
 [3]: STM
   version: 0000
   xlen: 32
 [4]: CTM
   version: 0000
   addr_width: 32
   data_width: 32
 [5]: DEM-UART
   version: 0000
Wait for connection
Connected
opensocdebugd: /home/ppenna/Documents/Projects/bluedragon/optimsoc/external/glip/src/backend_uart/sw/backend_uart.c:839: thread_func: Assertion `buffer[0] == 0xfe' failed.

Thread 2 "opensocdebugd" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffff6b86700 (LWP 15559)]
0x00007ffff761e428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.

Thread 3 (Thread 0x7ffff6385700 (LWP 15560)):
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x00007ffff73dc0fc in cbuf_wait_for_level_change (buf=0x6036c0, level=6) at /home/ppenna/Documents/Projects/bluedragon/optimsoc/external/glip/src/cbuf.c:714
#2  0x00007ffff73dd5f0 in gb_uart_read_b (ctx=0x6035f0, channel=<optimized out>, size=18, data=0x7ffff6384e62 "", size_read=0x7ffff6384e58, timeout=0) at /home/ppenna/Documents/Projects/bluedragon/optimsoc/external/glip/src/backend_uart/sw/backend_uart.c:578
#3  0x00007ffff7bd2232 in receiver_thread_function (arg=0x603460) at /home/ppenna/Documents/Projects/bluedragon/optimsoc/external/opensocdebug/software/src/connection-standalone.c:71
#4  0x00007ffff79ba6ba in start_thread (arg=0x7ffff6385700) at pthread_create.c:333
#5  0x00007ffff76f03dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 2 (Thread 0x7ffff6b86700 (LWP 15559)):
#0  0x00007ffff761e428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff762002a in __GI_abort () at abort.c:89
#2  0x00007ffff7616bd7 in __assert_fail_base (fmt=<optimized out>, assertion=assertion@entry=0x7ffff73e44b2 "buffer[0] == 0xfe", file=file@entry=0x7ffff73e42b8 "/home/ppenna/Documents/Projects/bluedragon/optimsoc/external/glip/src/backend_uart/sw/backend_uart.c", line=line@entry=839, function=function@entry=0x7ffff73e4530 <__PRETTY_FUNCTION__.5731> "thread_func") at assert.c:92
#3  0x00007ffff7616c82 in __GI___assert_fail (assertion=assertion@entry=0x7ffff73e44b2 "buffer[0] == 0xfe", file=file@entry=0x7ffff73e42b8 "/home/ppenna/Documents/Projects/bluedragon/optimsoc/external/glip/src/backend_uart/sw/backend_uart.c", line=line@entry=839, function=function@entry=0x7ffff73e4530 <__PRETTY_FUNCTION__.5731> "thread_func") at assert.c:101
#4  0x00007ffff73de898 in thread_func (ctx_void=0x6035f0) at /home/ppenna/Documents/Projects/bluedragon/optimsoc/external/glip/src/backend_uart/sw/backend_uart.c:839
#5  0x00007ffff79ba6ba in start_thread (arg=0x7ffff6b86700) at pthread_create.c:333
#6  0x00007ffff76f03dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 1 (Thread 0x7ffff7fb2740 (LWP 15555)):
#0  0x00007ffff79c387f in __libc_recv (fd=fd@entry=5, buf=buf@entry=0x614940, n=n@entry=2, flags=flags@entry=256) at ../sysdeps/unix/sysv/linux/x86_64/recv.c:28
#1  0x00000000004011f2 in recv (__flags=256, __n=2, __buf=0x614940, __fd=5) at /usr/include/x86_64-linux-gnu/bits/socket2.h:44
#2  main (argc=<optimized out>, argv=<optimized out>) at /home/ppenna/Documents/Projects/bluedragon/optimsoc/external/opensocdebug/software/src/tools/daemon/daemon.c:107
```

imphil, 2017-07-25:
Thanks for the backtrace. We actually crash in glip on an assertation (https://github.com/TUM-LIS/glip/blob/master/src/backend_uart/sw/backend_uart.c#L839), so it's either a transmission problem or a bug somewhere in the glip software or hardware. It's actually one of the more annoying things to debug and I'll need to reproduce it here in order to have a look at it.

@ppenna Until then, you can try working around that by changing the frequency of the transmission from 12 MBaud to something smaller. You can do that by changing the parameter in the compute_tile toplevel where glip_uart_toplevel is instantiated.

imphil, 2017-07-25:
@ppenna just to be sure: you hit that `assert` randomly after some time and after you have already transmitted gotten some data successfully? It's not a startup crash?

ppenna, 2017-07-25:
@imphil The random crash occurs after some data is transmitted, and it's sure to not be a startup crash. I'll try the workaround in the mean time. Ty

wallento, 2017-07-25:
Hi, the part with the assert looks suspicious to me. Let me have a look the next days..

wallento, 2017-08-22:
It seems it is the UART glip transport, which needs a full review. The path FPGA->Host seems to have a bug, which also explains a previously reported problem with verifying initialized memory.

imphil, 2017-08-30:
Opened https://github.com/TUM-LIS/glip/issues/47 for further tracking in GLIP.

imphil, 2017-11-21:
@ppenna I think we've found the bug in GLIP and I've updated the master branch of optimsoc with the required changes. Can you try if that fixes the problem for you? (You might need to merge the master branch into the host-tile branch if you're using that one.)

imphil, 2018-02-16:
closing as the bug should be fixed and didn't get any response @ppenna

closed 2018-02-16 by imphil