optimsoc/optimsoc #78: Make UART-DEM work in Linux tty layer

imphil, 2017-08-18:
UART-DEM currently works during the Linux bootup only when a trivial driver is used, but not if requests go through the TTY layer. 

Debugging results:
- All data which should be sent is written to an internal circular buffer, and then the transfer of data is triggered by calling `__start_tx()` in 8250_port.c
- All sending of data happens in an ISR triggered by the THRE interrupt (`serial8250_tx_chars()` in 8250_port.c)
- Currently UART-DEM doesn't send any interrupts and ignores the writes to the interrupt enable register and reads to the interrupt status register.

TODOs:
- [ ] Kernel: Adjust DTS to use the right interrupt line. I'm choosing 4 for now.
- [ ] OSD: Enhance UART-DEM to implement at least the TX interrupt part of the 16550 uart controllers.

ppenna, 2017-08-18:
@imphil sure!

ppenna, 2017-08-18:
@imphil done 3223854ffcd6aa026eb085fc4c6c9317e68faa3d

imphil, 2017-08-18:
Towards the second goal, I've pushed a proof-of-concept to the host-tile branch as 5aeb2018aa8049f53881419e5653ec856e8e4983. With that commit
- I see printf() messages from user space (hooray!)
- However, I also get messages `serial8250: too much work for irq4`. This is most likely caused by me just holding the interrupt line high whenever we can; it should probably be just a single edge!?

@wallento can you take over from here? I'm running out of time on this task.

TODOs:
- clean up the implementation to proper Verilog
- figure out right interrupt behavior

ppenna, 2017-08-18:
@imphil done https://github.com/ppenna/linux/commit/3223854ffcd6aa026eb085fc4c6c9317e68faa3d

imphil, 2017-08-18:
Just to show that it's working, thats how my terminal looks :)

[image]

ppenna, 2017-08-18:
@imphil I was wondering if there is another device hooked on IRQ 4

imphil, 2017-08-18:
no, irq4 was until now in optimsoc

stffrdhrn, 2017-08-18:
I would recommend keeping irq 2 as that's the standard for openrisc UART irqs.

stffrdhrn, 2017-08-19:
@imphil BTW, great work investigating and finding the difference between UART-DEM and the 8520 driver.  It is possible to just use the 8520 core we usually use? 

https://github.com/olofk/uart16550/tree/master/rtl/verilog

imphil, 2017-08-20:
@wallento wrote the original code and said he will also look at the big over the weekend. I also thought about using the existing uart implementation and extending it towards the noc interface. Especially I expect the next bugs to come up when we talk about input and not output streams as well. I however don't know what the exact motivation for the current design was, is I'll leave this for Stefan.

wallento, 2017-08-20:
The choice was for simplicity and I don't like the standard uart core code quality too much.. I don't think there is much of a difference with respect to the necessary work.

wallento, 2017-08-20:
Finally, we don't want LGPL code in our codebase I assume ;)

imphil, 2017-08-22:
With the latest changes in the host-tile branch by @wallento I get user-space output on the VCU108 (great!), but it seems we drop characters sometimes.

[image]

In this screenshot, Look at the third line from the bottom: `This architecture does not`. This message, coming from the kernel, is truncated by the subsequent user-space output `initializing noc device`. @wallento could it be that the ready signal is not properly lowered when there is still a transfer ongoing?

imphil, 2017-08-22:
Actually we loose data also from the kernel messages. After "libphy: Fixed ...." there should be a message "NET: Registered protocol family 17" (see previous [screenshot](https://github.com/optimsoc/optimsoc/issues/78#issuecomment-323227623)). Both are kernel messages.

imphil, 2017-08-22:
STDIN also works now from my initial testing. Nice work @wallento !

wallento, 2017-08-22:
Out of curiosity @imphil, are you using the CTM too?

imphil, 2017-08-22:
yes, also using CTM

imphil, 2017-08-22:
To answer the question which might have been implied:
- High traffic scenarios on the Debug NoC should not lead to any lost packages for the UART-DEM, as we can have backpressure here (in contrast to CTM).
- Before the rewrite, we didn't loose any data in the same traffic scenario (same CTM and STM traces enabled).

wallento, 2017-08-23:
I cannot confirm the issue on the Nexys4 Board. Can you please check once without CTM to guide me. Your points are right, but the backpressure got touched during the rework and I suspect it may lie there.

imphil, 2017-08-23:
I'm running a synthesis with ILA to test and will get back to you.

imphil, 2017-08-23:
Debugging shows that inside the module osd_dem_uart_16550 the backpressure from the NoC sending part isn't handled properly. This manifests in `out_ready == 0` and `out_valid == 1` at the same time, causing dropped characters.

@wallento How should backpressure through wishbone be handled? Simply don't set the wb_ack signal to high until the data word was really consumed? (Even better would be a patch of course :-) )

imphil, 2018-10-30:
With the latest merge from Open SoC Debug Linux now boots up to userspace with all stdout being forwarded to the host.

closed 2018-10-30 by imphil