optimsoc/optimsoc #93: Evaluate Kactus2 to create structural Verilog modules

imphil, 2017-10-16:
Currently, we have a couple of purely structural Verilog files in OpTiMSoC: the toplevel files at various levels of the hierarchy. Creating these files is tedious and error-prone (getting wiring wrong in Verilog is pretty easy as it relies mostly on copy/paste). At the same time, manually creating these files makes experimentation on higher abstraction levels harder than it needs to be.

In order to solve these issues, we're looking at a graphical tool to create structural Verilog using a block level design method. Kactus2 seems to be an obvious candidate in this respect.

With raising complexity, these three files could be generated by Kactus2:
- The system toplevel (e.g. https://github.com/optimsoc/optimsoc/blob/acf16404eb4cad43b7a2d6dd699f9d330ad1d41a/src/soc/hw/system_2x2_cccc_dm/verilog/system_2x2_cccc_dm.sv)
- The board-specific "real" top level (e.g. https://github.com/optimsoc/optimsoc/blob/master/examples/fpga/nexys4ddr/compute_tile/rtl/verilog/compute_tile_dm_nexys4.sv)
- The tile toplevel, representing a single tile (e.g. https://github.com/optimsoc/optimsoc/blob/086cd8af3484ec6772f5f4ff6790ae983d7529e3/src/soc/hw/compute_tile_dm/verilog/compute_tile_dm.sv). The challenge here is: how to represent a bus with a flexible number of masters and slaves?

In order to proceed with the evaluation of using Kactus2 in these cases, we should:

- [ ] Create IPXact descriptions of all modules included in the mentioned toplevels, starting with all modules used in the system_2x2_cccc_dm file
- [ ] Check the output generated by Kactus2 when trying to re-create the structural Verilog file (e.g. system_2x2_cccc_dm).

Along the way, we need to figure out how/if we want/need to interface with fusesoc, or if we can live with the duplication of file lists in both description formats. @olofk might have some ideas in this regard as well  -- but let's discuss them when we stumble on specific problems.

wallento, 2017-10-18:
I like the idea and contrary to our prior approaches using IP-XACT and Kactus seems more solid. One of the issues I could think about is parameterization, but lets see.

olofk, 2017-10-18:
Yes, I have some ideas and perhaps some infrastructure that could come in use, such as bus interfaces and models of a few cores. Looking at the files you linked to, I see you're using a lot of parametrization to set the number of instances to use for a certain component. This is one area where IP-XACT is not very good.

Another type of component that might need special handling is bus interconnect (I believe it's wb_bus_b3 in this case). My solution for this in wb_intercon has been to generate a static component from a simple configuration file. wb_intercon currently only generates the verilog code, but I wrote an IP-XACT component generator to go along with it some years ago, but never checked it in.

When it comes to using the IP-XACT files in FuseSoC, it's just to specify the component files and FuseSoC will pick up the source files. Unfortunately there is currently no way to specify scope and usage for files from IP-XACT components. I have some rough ideas here, but hasn't been a priority