optimsoc/optimsoc #104: Setup time violation @ 100 MHz starting at noc_buffer.out_valid

koenenwmn, 2018-02-19:
When changing the logic clock to 100 MHz we get the following timing error. It's essentially a very long path of the valid signal starting at `u_na/u_mpbuffer/u_buffer/genblk2[1].u_endpoint/u_packetbuffer_out/out_valid` and leading up to the valid signal of the MIG AXI.

@wallento do you have an idea how to resolve that one?

[schematic.pdf](https://github.com/optimsoc/optimsoc/files/1737166/schematic.pdf)

~~~
--------------------------------------------------------------------------------------
Slack (VIOLATED) :        -1.788ns  (required time - arrival time)
  Source:                 u_compute_tile/u_na/u_mpbuffer/u_buffer/genblk2[1].u_endpoint/u_packetbuffer_out/data_last_buf_reg[5]/C
                            (rising edge-triggered cell FDRE clocked by clk  {rise@0.000ns fall@3.332ns period=9.996ns})
  Destination:            u_board/u_ddr4_c0_axi_clk_converter/inst/gen_clock_conv.gen_sync_conv.gen_conv_write_ch.w_sync_clock_converter/gen_sync_clock_converter.m_tvalid_r_reg/D
                            (rising edge-triggered cell FDRE clocked by mmcm_clkout0  {rise@0.000ns fall@1.666ns period=3.332ns})
  Path Group:             mmcm_clkout0
  Path Type:              Setup (Max at Slow Process Corner)
  Requirement:            3.332ns  (mmcm_clkout0 rise@3.332ns - clk rise@0.000ns)
  Data Path Delay:        4.557ns  (logic 1.523ns (33.421%)  route 3.034ns (66.579%))
  Logic Levels:           14  (LUT2=1 LUT3=1 LUT4=1 LUT5=5 LUT6=6)
  Clock Path Skew:        -0.568ns (DCD - SCD + CPR)
    Destination Clock Delay (DCD):    4.436ns = ( 7.768 - 3.332 ) 
    Source Clock Delay      (SCD):    4.731ns
    Clock Pessimism Removal (CPR):    -0.273ns
  Clock Uncertainty:      0.054ns  ((TSJ^2 + DJ^2)^1/2) / 2 + PE
    Total System Jitter     (TSJ):    0.071ns
    Discrete Jitter          (DJ):    0.082ns
    Phase Error              (PE):    0.000ns
  Clock Net Delay (Source):      2.748ns (routing 0.969ns, distribution 1.779ns)
  Clock Net Delay (Destination): 2.445ns (routing 0.892ns, distribution 1.553ns)

    Location             Delay type                Incr(ns)  Path(ns)    Netlist Resource(s)
  -------------------------------------------------------------------    -------------------
                         (clock clk rise edge)        0.000     0.000 r  
    G31                                               0.000     0.000 r  sysclk1_300_p (IN)
                         net (fo=0)                   0.000     0.000    u_board/u_mig_ultrascale/inst/u_ddr4_mem_intfc/u_mig_ddr4_phy/inst/u_ddr4_phy_pll/diff_input_clk.u_ibufg_sys_clk/I
    HPIOBDIFFINBUF_X0Y156
                         DIFFINBUF (Prop_DIFFINBUF_HPIOBDIFFINBUF_DIFF_IN_P_O)
                                                      0.537     0.537 r  u_board/u_mig_ultrascale/inst/u_ddr4_mem_intfc/u_mig_ddr4_phy/inst/u_ddr4_phy_pll/diff_input_clk.u_ibufg_sys_clk/DIFFINBUF_INST/O
                         net (fo=1, routed)           0.090     0.627    u_board/u_mig_ultrascale/inst/u_ddr4_mem_intfc/u_mig_ddr4_phy/inst/u_ddr4_phy_pll/diff_input_clk.u_ibufg_sys_clk/OUT
    G31                  IBUFCTRL (Prop_IBUFCTRL_HPIOB_I_O)
                                                      0.000     0.627 r  u_board/u_mig_ultrascale/inst/u_ddr4_mem_intfc/u_mig_ddr4_phy/inst/u_ddr4_phy_pll/diff_input_clk.u_ibufg_sys_clk/IBUFCTRL_INST/O
                         net (fo=2, routed)           0.865     1.492    u_board/u_mig_ultrascale/inst/u_ddr4_infrastructure/mmcm_clk_in
    MMCME3_ADV_X0Y6      MMCME3_ADV (Prop_MMCME3_ADV_CLKIN1_CLKOUT0)
                                                     -0.231     1.261 r  u_board/u_mig_ultrascale/inst/u_ddr4_infrastructure/gen_mmcme3.u_mmcme_adv_inst/CLKOUT0
                         net (fo=3, routed)           0.440     1.701    u_board/lopt
    BUFGCE_DIV_X0Y25     BUFGCE_DIV (Prop_BUFGCE_DIV_I_O)
                                                      0.282     1.983 r  u_board/BUFGCE_DIV_CLK1_inst/O
    X1Y5 (CLOCK_ROOT)    net (fo=5377, routed)        2.748     4.731    u_compute_tile/u_na/u_mpbuffer/u_buffer/genblk2[1].u_endpoint/u_packetbuffer_out/clk
    SLICE_X74Y298        FDRE                                         r  u_compute_tile/u_na/u_mpbuffer/u_buffer/genblk2[1].u_endpoint/u_packetbuffer_out/data_last_buf_reg[5]/C
  -------------------------------------------------------------------    -------------------
    SLICE_X74Y298        FDRE (Prop_EFF2_SLICEL_C_Q)
                                                      0.116     4.847 f  u_compute_tile/u_na/u_mpbuffer/u_buffer/genblk2[1].u_endpoint/u_packetbuffer_out/data_last_buf_reg[5]/Q
                         net (fo=5, routed)           0.250     5.097    u_compute_tile/u_na/u_mpbuffer/u_buffer/genblk2[1].u_endpoint/u_packetbuffer_out/data_last_buf[5]
    SLICE_X72Y297        LUT5 (Prop_D6LUT_SLICEL_I4_O)
                                                      0.175     5.272 f  u_compute_tile/u_na/u_mpbuffer/u_buffer/genblk2[1].u_endpoint/u_packetbuffer_out/FSM_onehot_state[8]_i_17/O
                         net (fo=2, routed)           0.190     5.462    u_compute_tile/u_na/u_mpbuffer/u_buffer/genblk2[1].u_endpoint/u_packetbuffer_out/FSM_onehot_state[8]_i_17_n_0
    SLICE_X72Y298        LUT6 (Prop_C6LUT_SLICEL_I5_O)
                                                      0.070     5.532 f  u_compute_tile/u_na/u_mpbuffer/u_buffer/genblk2[1].u_endpoint/u_packetbuffer_out/FSM_onehot_state[8]_i_8/O
                         net (fo=1, routed)           0.152     5.684    u_compute_tile/u_na/u_mpbuffer/u_buffer/genblk2[1].u_endpoint/u_packetbuffer_out/FSM_onehot_state[8]_i_8_n_0
    SLICE_X72Y299        LUT6 (Prop_F6LUT_SLICEL_I0_O)
                                                      0.174     5.858 r  u_compute_tile/u_na/u_mpbuffer/u_buffer/genblk2[1].u_endpoint/u_packetbuffer_out/FSM_onehot_state[8]_i_6/O
                         net (fo=8, routed)           0.215     6.073    u_compute_tile/u_na/u_mux1/rp_reg[0]_0
    SLICE_X69Y299        LUT6 (Prop_F6LUT_SLICEL_I2_O)
                                                      0.072     6.145 f  u_compute_tile/u_na/u_mux1/FSM_onehot_state[8]_i_4/O
                         net (fo=7, routed)           0.347     6.492    u_compute_tile/u_na/u_dma/target/active_reg[0]_0
    SLICE_X69Y300        LUT6 (Prop_E6LUT_SLICEL_I2_O)
                                                      0.040     6.532 r  u_compute_tile/u_na/u_dma/target/wb_arb[1]_i_2/O
                         net (fo=2, routed)           0.229     6.761    u_compute_tile/u_na/u_dma/ctrl_initiator/u_wbreq/wb_target_cyc_o
    SLICE_X69Y300        LUT4 (Prop_D6LUT_SLICEL_I3_O)
                                                      0.132     6.893 r  u_compute_tile/u_na/u_dma/ctrl_initiator/u_wbreq/prev_grant[2]_i_2/O
                         net (fo=4, routed)           0.151     7.044    u_compute_tile/u_bus/u_mux/busms_cyc_o_flat[0]
[... 31 lines cut ...]
    HPIOBDIFFINBUF_X0Y156
                         DIFFINBUF (Prop_DIFFINBUF_HPIOBDIFFINBUF_DIFF_IN_P_O)
                                                      0.380     3.712 r  u_board/u_mig_ultrascale/inst/u_ddr4_mem_intfc/u_mig_ddr4_phy/inst/u_ddr4_phy_pll/diff_input_clk.u_ibufg_sys_clk/DIFFINBUF_INST/O
                         net (fo=1, routed)           0.051     3.763    u_board/u_mig_ultrascale/inst/u_ddr4_mem_intfc/u_mig_ddr4_phy/inst/u_ddr4_phy_pll/diff_input_clk.u_ibufg_sys_clk/OUT
    G31                  IBUFCTRL (Prop_IBUFCTRL_HPIOB_I_O)
                                                      0.000     3.763 r  u_board/u_mig_ultrascale/inst/u_ddr4_mem_intfc/u_mig_ddr4_phy/inst/u_ddr4_phy_pll/diff_input_clk.u_ibufg_sys_clk/IBUFCTRL_INST/O
                         net (fo=2, routed)           0.778     4.541    u_board/u_mig_ultrascale/inst/u_ddr4_infrastructure/mmcm_clk_in
    MMCME3_ADV_X0Y6      MMCME3_ADV (Prop_MMCME3_ADV_CLKIN1_CLKOUT0)
                                                      0.335     4.876 r  u_board/u_mig_ultrascale/inst/u_ddr4_infrastructure/gen_mmcme3.u_mmcme_adv_inst/CLKOUT0
                         net (fo=3, routed)           0.372     5.248    u_board/u_mig_ultrascale/inst/u_ddr4_infrastructure/mmcm_clkout0
    BUFGCE_X0Y158        BUFGCE (Prop_BUFCE_BUFGCE_I_O)
                                                      0.075     5.323 r  u_board/u_mig_ultrascale/inst/u_ddr4_infrastructure/u_bufg_divClk/O
[... cut ...]
                                                      0.059     7.500    u_board/u_ddr4_c0_axi_clk_converter/inst/gen_clock_conv.gen_sync_conv.gen_conv_write_ch.w_sync_clock_converter/gen_sync_clock_converter.m_tvalid_r_reg
  -------------------------------------------------------------------
                         required time                          7.500    
                         arrival time                          -9.288    
  -------------------------------------------------------------------
                         slack                                 -1.788    

~~~

imphil, 2018-03-12:
To avoid working on code which will be removed soon anyways we'll revisit this after @wallento merged his improved NA implementation. (see discussion in #112)

wallento, 2018-03-13:
Hi, I am not fully sure if it is covered by my patch set. Anyhow, lets revisit it once my stuff is merged. Cheers

imphil, 2018-03-15:
With a higher utilization I get a very similar timing violation also in a Nexys 4 DDR design system_2x2_cccc @ 50 MHz, hence we should invest a bit more time removing this critical path.

@koenenwmn could you try to spend some time on figuring out where exactly the problem lies? All critical paths somehow go through the packet buffer in mpbuffer, somewhere there I'd expect we need to insert a register stage somewhere around there.