optimsoc/optimsoc #112: Circular loop through mpbuffer

imphil, 2018-03-12:
Verilator reports multiple combinatorial loops, many of them going through the network adapter and message passing units.

@wallento do you have had a look at these before or do you have changes laying around which would make us looking at these warnings a waste of time (as it's "almost dead" code anyways)?

~~~
%Warning-UNOPTFLAT: ../src/optimsoc_networkadapter_networkadapter_ct_0/verilog/networkadapter_ct.sv:70: Signal unoptimizable: Feedback to clock or circular logic: tb_compute_tile.u_compute_tile.__Vcellout__u_na__wbm_bte_o
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_networkadapter_networkadapter_ct_0/verilog/networkadapter_ct.sv:70:  tb_compute_tile.u_compute_tile.__Vcellout__u_na__wbm_bte_o
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:585:  ASSIGNW
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:152:  tb_compute_tile.u_compute_tile.busms_bte_o
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:219:  ASSIGNW
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:189:  tb_compute_tile.u_compute_tile.busms_bte_o_flat
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_wb_interconnect_mux_0/verilog/wb_mux.sv:156:  ALWAYS
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_wb_interconnect_bus_0/verilog/wb_bus_b3.v:172:  tb_compute_tile.u_compute_tile.u_bus.bus_we
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_wb_interconnect_decode_0/verilog/wb_decode.sv:207:  ALWAYS
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:200:  tb_compute_tile.u_compute_tile.bussl_we_i_flat
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:232:  ASSIGNW
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:163:  tb_compute_tile.u_compute_tile.bussl_we_i
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:607:  ASSIGNW
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_networkadapter_networkadapter_ct_0/verilog/networkadapter_ct.sv:81:  tb_compute_tile.u_compute_tile.__Vcellinp__u_na__wbs_we_i
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_wb_interconnect_decode_0/verilog/wb_decode.sv:207:  ALWAYS
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_networkadapter_networkadapter_ct_0/verilog/networkadapter_ct.sv:128:  tb_compute_tile.u_compute_tile.u_na.wbif_we_i
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_mp_buffer_0/verilog/mpbuffer.sv:88:  ASSIGNW
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_mp_buffer_0/verilog/mpbuffer.sv:65:  tb_compute_tile.u_compute_tile.u_na.u_mpbuffer.u_buffer.bus_ack_mod
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_networkadapter_networkadapter_ct_0/verilog/networkadapter_ct.sv:211:  ASSIGNW
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_networkadapter_networkadapter_ct_0/verilog/networkadapter_ct.sv:132:  tb_compute_tile.u_compute_tile.u_na.wbif_ack_o
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_wb_interconnect_decode_0/verilog/wb_decode.sv:199:  ALWAYS
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_networkadapter_networkadapter_ct_0/verilog/networkadapter_ct.sv:88:  tb_compute_tile.u_compute_tile.__Vcellout__u_na__wbs_dat_o
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:589:  ASSIGNW
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:170:  tb_compute_tile.u_compute_tile.bussl_dat_o
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:239:  ASSIGNW
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:207:  tb_compute_tile.u_compute_tile.bussl_dat_o_flat
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_wb_interconnect_decode_0/verilog/wb_decode.sv:199:  ALWAYS
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_wb_interconnect_bus_0/verilog/wb_bus_b3.v:176:  tb_compute_tile.u_compute_tile.u_bus.bus_rdat
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_wb_interconnect_mux_0/verilog/wb_mux.sv:166:  ALWAYS
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:193:  tb_compute_tile.u_compute_tile.busms_dat_i_flat
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:223:  ASSIGNW
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:156:  tb_compute_tile.u_compute_tile.busms_dat_i
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_tile_compute_tile_dm_0/verilog/compute_tile_dm.sv:601:  ASSIGNW
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_networkadapter_networkadapter_ct_0/verilog/networkadapter_ct.sv:74:  tb_compute_tile.u_compute_tile.__Vcellinp__u_na__wbm_dat_i
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_lisnoc_dma_0/rtl/dma/lisnoc_dma.sv:316:  ALWAYS
%Warning-UNOPTFLAT:      Example path: ../src/optimsoc_networkadapter_networkadapter_ct_0/verilog/networkadapter_ct.sv:70:  tb_compute_tile.u_compute_tile.__Vcellout__u_na__wbm_bte_o
~~~

wallento, 2018-03-12:
Hi,

I can have a look at it, but as you say this part is reworked in my private branch and I can push it earlier than the remaining part. I am also not sure if this is some other nasty false path, because verilator sometimes shows wrong paths for arrays of signals.

ETA is the next two weeks.

Cheers,
Stefan

imphil, 2018-03-12:
OK, then we'll leave this as-is for now until you push your changes. If you have parts which can go to master earlier that would help on my side to avoid working on stuff which is in the process of being replaced (and probably would also reduce the stack of patches your sitting on to a more maintainable level)