optimsoc/optimsoc #122: VCU108 FX3 system broken

wallento, 2018-03-31:
Hi,

I have an issue with the VCU108 and the FX3. Generally, OSD works, but once I start initializing the memory I have memory losses.

Steps to reproduce:
- The issue is on the current master (cd674ac), for convenience I created a branch with the relevant debug inserted: https://github.com/optimsoc/optimsoc/tree/bug-vcu108fx3
- Build the system on this branch: `fusesoc --cores-root /path/to/fpga/vcu108/system_2x2_cccc/ build  optimsoc:examples:system_2x2_cccc_vcu108 --HOST_IF=usb3`
- Download bitstream and set debug trigger to the mam strobe signal (0->1)
- Run: `osd-target-run -e hello.elf -b cypressfx3`

Here is what OSD is sending when it starts initializing the memory:

```
packet 1:
0x0001
0x0401
0x8000
0xc0ff
0x0000
0x0000
0x0000
0x0000

packet 2:
0x0001
0x0401
0x8000
0x0000
0x0000
0x0000
0x0000
0x0000

packet 3:
0x0001
0x0401
0x8000
0x0000
0x0000
0x0000
0x0000
0x0000
```

This is the data I see on the glip interface:

```
0x0008
0x0001
0x0401
0x8000
0xc0ff
0x0000
0x0000
0x0000
0x0000
0x0008
0x0001
0x0401
0x8000
0x0000
0x0000
0x0000
0x0000
0x0000
0x0008
0x0001
0x8000
0x0000
0x0000
0x0008 (<- wrong from here)
0x0401
0x0000
0x0000
[...]
```

So, it looks a bit that there is some data loss and the fourth packet's header slips into the third packet's payload. After that it hangs then.

@koenenwmn I think it makes most sense if you can have a look at this.

imphil, 2018-04-03:
@koenenwmn maybe a good first step would be to revert back to the older logic toplevel (i.e. the one with the timing issues) to see if the problem came from there.

imphil, 2018-04-21:
@wallento: fyi @koenenwmn has found the bug and is working on a fix, ETA next week.

imphil, 2018-04-21:
Ah, forgot to mention: it's a issue in the GLIP cypressfx3 backend, as a workaround using the uart backend will work.