optimsoc/optimsoc #125: Integrate Ethernet work by Dominik

imphil, 2018-04-23:
The data is available (inside LIS) at /nfs/projects/optimsoc/aramis/

wallento, 2018-04-23:
Hi,

before you start on this @koenenwmn, can you please elaborate on what the contribution is with respect to hardware requirements and software stack?

In particular, a few questions are interesting for me:

 - Does it depend on non-free IP from Xilinx?
 - Does it work on all targeted Xilinx-based FPGA boards?
 - Does it smoothly integrate with a compute tile? Hence, can it become like part of the host tile?

Cheers,
Stefan

imphil, 2018-04-23:
This issue is about the very narrow issue of cleaning up/integrating the solution developed by Dominik for the purposes of his demo. It's targeting only the VCU108 board, works only on a single compute tile design and makes use of significant portions of Xilinx IP.

imphil, 2018-04-23:
We'll need to see how much of it we want to integrate into upstream OpTiMSoC by default, and how much of it should be specific for the demo project of the Aramis project.

wallento, 2018-04-23:
okay, I see, it may make sense to keep that on gitlab.lrz.de for the first phase, private discussion pending.