optimsoc/optimsoc #134: Interrupt for DEM UART

imphil, 2018-08-20:
Linux assumes by default that the UART interrupt is on line 2 (https://elixir.bootlin.com/linux/latest/source/arch/openrisc/boot/dts/or1ksim.dts#L47).

In OpTiMSoC we have the NA DMA interrupt on line 2 at the moment. However, a quick grep showed that no software actually uses this interrupt line. 

How should we proceed @wallento? Map UART to a non-standard interrupt in OpTiMSoC, or move the NA DMA interrupt to line 4?

wallento, 2018-08-20:
I was kind of sure, we faced this issue before and fixed it. Let me dig into the various branches, will get back asap.

imphil, 2018-08-20:
We had UART initially on interrupt 4 in the host-tile branch and switched back to interrupt 2. Not sure what we did with the DMA interrupt, however.

imphil, 2018-09-03:
With https://github.com/optimsoc/optimsoc/pull/136 the DMA interrupt now moved to 4, MP is where it was and UART is at 2. I didn't see any software references to the DMA interrupt, maybe my grep was just wrong, or we never used it.

closed 2018-09-03 by imphil