optimsoc/optimsoc #135: Undefined reference at make

ticksmas, 2018-08-27:
Hi there!
When I tried to "make" from the optimsoc-sources folder, I got the following error during the build:

`./optimsoc-sources/external/opensocdebug/software/src/tools/osd-device-gateway/osd-device-gateway.c:82: undefined reference to `glip_parse_option_string'
`
I have found the headers, and tried to include it in osd-device-gateway, without success. As a workaround, I copied the function code inside this file, but the error came in other.

I am using Ubuntu 16.04 LTS, kernel 4.4.0-83-generic.
Any tips to solve this?

imphil, 2018-08-27:
Are you sure that you have a clean source tree, equal to the latest upstream version (the master branch)? Please also ensure that you don't have the OPTIMSOC environment variable set when building OpTiMSoC.

ticksmas, 2018-08-27:
The OPTIMSOC environment variable was set. 
I'll try from scratch, without installing the binaries.
Thank you!

closed 2018-08-27 by ticksmas

imphil, 2018-08-27:
Looks like we really should get around to implement https://github.com/optimsoc/optimsoc/issues/34 :-) If you have time, I'm happy to have a look at a PR (it should not require more than two lines in tools/build.py)