optimsoc/optimsoc #144: clog2 fails with Vivado Versions < 2018.x after commit 271bad18f0aecc675509171788b07fdcef935b7c

bergyla, 2018-10-03:
Using clog2 (as called by clog2_width in optimsoc_functions.sv  - used in mpbuffer_endpoint.sv in ) from Vivado will fail with Vivado Versions < 2018.x:
Vidado Error: "clog2_width is not a valid constant function call"

This brakes compatibility to the advised 2016.4 as recomended in the actual documentation.

Btw: Using Vivado 2018.2 as announced in Commit note will require > 4GB Memory since the built script will fail with memory allocation error after some while in FPGA Synthesis process

wallento, 2018-10-03:
Thanks for reporting. @imphil, can you have a look at reverting and fixing the warning instead?

bergyla, 2018-10-04:
Vivado 2018.2.1 (Service Release of the struggling Vivado Release cuased Philip to provide the change) will run well with modified build.py Fix (as provided in fix-version-handling branch) with the original code of optimsoc_functions.sv.

So from this side, there is no more need to use the built in vlog2 Function but to install the Release Update...

*Xilinx released a 2nd Service Release 2018.2 (Vivado 2018.2.2).
I will rerun the built run and report accordingly...

imphil, 2018-10-04:
Thanks Andreas for reporting this. I've now reverted the commit. This should resolve the error at the cost of a warning, which is a reasonable thing to do. Let's wait until the dust settles in this area of Vivado and then try again to find a more permanent fix.

commit 035f3a51 (2018-10-04): Revert "Remove non-SV clog2() function"

This reverts commit 271bad18f0aecc675509171788b07fdcef935b7c.

It was reported that Vivado 2018.2.x fixed this problem, and more
importantly, Vivado versions before 2018 don't see $clog2() as a
constant function, which causes a synthesis error (as opposed to fixing
a warning).

Let's revert this for now, and see how things evolve.

Fixes #144
closed 2018-10-04 by imphil