optimsoc/optimsoc #146: Following fusesoc simulation Examples failing

bergyla, 2018-10-05:
Following the 1.4.1 Building Hardware fails on the the newly added Parameters: CORE_ENABLE_FPU + CORE_ENABLE_PERFCOUNTERS
BASE_CONFIG is mismatching:

/opt/optimsoc/prebuilt/verilator-3.902/bin/verilator -f optimsoc_examples_compute_tile_sim_0.vc --trace -Wno-fatal -CFLAGS "-std=c++11" -LDFLAGS "-pthread" -Wwarn-lint
%Error: Assignment pattern missed initializing elements: MEMBERDTYPE 'CORE_ENABLE_FPU'
%Error: Assignment pattern missed initializing elements: MEMBERDTYPE 'CORE_ENABLE_PERFCOUNTERS'
%Warning-WIDTH: ../src/optimsoc_examples_compute_tile_sim_0/tb_compute_tile.sv:56: Operator VAR 'BASE_CONFIG' expects 1581 bits on the Initial value, but Initial value's CONCAT generates 1579 bits.

Tested with master on 035f3a519dba4d338a8ae2d13990f51fd6f899c5

imphil, 2018-10-09:
I'm not sure what steps create this problem for you, could you please add more information what you did/do to reach this error message?

imphil, 2018-12-18:
@bergyla Is this still a problem for you? Otherwise please close this issue or I'll close it within a week.

closed 2019-01-23 by imphil