optimsoc/optimsoc #154: Reenable system_2x2_cccc on Nexys 4 DDR

imphil, 2018-11-23:
Commit  94c76b2 disabled the 2x2 design on Nexys since it's too large. Investigate how to shrink it.

lucasandradecb, 2019-11-14:
Hello, I'm Lucas Andrade and I'm working with @ppenna  on a [Nanvix](https://github.com/nanvix/optimsoc) project that is portable to the OptimSoc platform.

When working with your OptimSoc master project, I identified that the system_2x2_cccc_nexys4ddr design was disabled.

However I found that when using Vivado WebPack 2019.1 and the command _**'make BUILD_EXAMPLES = yes BUILD_EXAMPLES_FPGA = yes BUILD_DOCS = no'**_ the system_2x2_cccc_nexys4ddr design works correctly, even generating the bitstream. In addition, I validated the bitstream on the Nexys4DDR card, running hello.elf provided by you, and it worked correctly.

By the executions and analyzes performed by me, this too large bug for the device has been resolved by using the new version of Vivado, which performs design build optimization that allows it to run on the board.

I hope I can add is my contribution to the project.

Best Regards,
Lucas Andrade

imphil, 2019-11-14:
Thanks for confirming that it works now @lucasandradecb!

Most likely this was fixed by a updated mor1kx version in https://github.com/optimsoc/optimsoc/commit/82331f854f7aad11278716c008f388370641bc71 (RAM wasn't inferred properly). 

I did a PR in #183 to re-enable that build.

closed 2019-11-14 by imphil