optimsoc/optimsoc #155: make fails on FPGA build with fusesoc 1.8.4 parse error on commit 71e74d7

bergyla, 2018-12-13:
make fails in FPGA Build process of nexys4ddr with the following Log-Messages: 

WARNING: Parse error. Ignoring file /optimsoc-sources/examples/fpga/nexys4ddr/compute_tile/compute_tile_nexys4ddr.core: Unknown item 'rule_parameters in section Spyglass'
ERROR: 'optimsoc:examples:compute_tile_nexys4ddr' or any of its dependencies requires 'compute_tile_nexys4ddr', but this core was not found

Removing the rule_parameters Section fix the issue.
fusesoc was built directly from GitHub reporting (latest) Version 1.8.4

imphil, 2018-12-13:
The current `master` branch of optimsoc depends on a fusesoc version newer than 1.8.4, which is not released yet (see https://github.com/optimsoc/optimsoc/blob/master/requirement_versions.yml#L8). 

If you don't want to use Spyglass simply commenting out the relevant spyglass lines should do the trick.

If you want to use spyglass, and until a newer release of fusesoc has happened, please use the "master" branch of fusesoc. You can install it quite easily:

```sh
git clone https://github.com/olofk/fusesoc.git
pip3 install --user .
# if you use --user, make sure that ~/.local/bin is in your PATH
fusesoc --version
# should give you something like
# $ fusesoc --version
# 1.8.5.dev22+g89b4ad5
```
Notes:
- Use pip or pip3 depending on what Python version you've been using before.
- Use `sudo pip` or `pip --user` depending on if you want to install fusesoc globally or only for your current user. Ideally use the same method as you used before to avoid ending up with two fusesoc versions installed. Otherwise, uninstall fusesoc first.

bergyla, 2018-12-13:
Work well. Thank you

closed 2018-12-13 by bergyla