optimsoc/optimsoc #158: Remove bin2vmem

imphil, 2019-01-03:
It seems that objcopy has a "verilog" target option which creates vmem files. See if that option works for us and remove bin2vmem in this case.

wallento, 2019-01-04:
I remember there were some issues before and I can find those mentions:

https://github.com/sifive/elf2hex/issues/1

https://sourceware.org/bugzilla/show_bug.cgi?id=19921

I will have a look into it!

imphil, 2019-01-04:
The patch in the sourceware bug is indeed what's needed, should just get that submitted. I'll give it a try on RISC-V Little Endian early next week to ensure that the little endian stuff works as well.

imphil, 2019-01-04:
(The other problem mentioned in sifive/elf2hex#1 isn't actually a problem for us.)