optimsoc/optimsoc #161: Dependency Problems while building with FUSESOC for VCU108

SRQ91, 2019-01-23:
I am trying to get the file structure for single tile quad core system of OpTiMSoC. The make of OpTiMSoC runs fine but generates only single core single tile vivado project in the objdir/examples/fpga/vcu108. A quadcore bit stream is generated but not the vivado project. To get this I go to examples/fpga/vcu108 and change the num of cores in .sv computetile file for 4. When trying to run this with fusesoc as mentioned in the readme, I get the following errors:

```
ERROR: Conflicting requirements:
Requirements: 'optimsoc_examples_system_2x2_cccc_vcu108 == 0-0' <- 'glip_backend_cypressfx3 >= 0-0'
    +optimsoc_examples_system_2x2_cccc_vcu108-0-0 was ignored because it depends on missing packages
Requirements: 'optimsoc_examples_system_2x2_cccc_vcu108 == 0-0'
    Install command rule (+optimsoc_examples_system_2x2_cccc_vcu108-0-0)

Failed to resolve dependencies for optimsoc:examples:system_2x2_cccc_vcu108
```

I am not sure why the make works fine but the fusesoc build method does not work.

imphil, 2019-01-23:
It looks like you didn't load the optimsoc-environment.sh file into your environment. Please have a look at the tutorial.

If you want to start your own project based on optimsoc, use one of the example projects as starting point, *copy* it into a new directory somewhere outside the optimsoc source directory, and start modifying it. You do not need to rebuild optimsoc itself, build and install it (or use a pre-built version), and then you only need to use fusesoc to build your own modified hardware.

The documentation at https://www.optimsoc.org/docs/master/user_guide/develop.html has more information on that.

If you only want to change the number of cores per tile in the `system_2x2_cccc_vcu108` example system you can do so by passing the --NUM_CORES=2 parameter to fusesoc: `fusesoc --cores-root $OPTIMSOC_SRC/examples build optimsoc:examples:compute_tile_vcu108 --HOST_IF usb3 --NUM_CORES 2`.

Or if you want to have a system with a single compute tile containing 8 cores you can do `fusesoc --cores-root $OPTIMSOC_SRC/examples build optimsoc:examples:system_2x2_cccc_vcu108 --HOST_IF usb3 --NUM_CORES 8`

SRQ91, 2019-01-23:
Thank you for the help. Fusesoc was not about to file the additional cores due to incorrect configuration of $FUSESOC_CORES.

closed 2019-01-23 by SRQ91