optimsoc/optimsoc #162: NUM_CORES not changing number of cores

SRQ91, 2019-01-23:
I am using this command to generate vivado project using FUSESOC:

`fusesoc --cores-root $OPTIMSOC_SRC/examples/fpga/vcu108/compute_tile build optimsoc:examples:compute_tile_vcu108 --NUM_CORES 2`

Fusesoc works to generate a build folder but only one core is instantiated in the compute_tile file. During execution Fusesoc shows this message:

```
source optimsoc_examples_compute_tile_vcu108_0.tcl
# create_project optimsoc_examples_compute_tile_vcu108_0
# set_property part xcvu095-ffva2104-2-e [current_project]
# set_property generic {NUM_CORES=2 } [get_filesets sources_1]
```

I am not sure what is the reason for this. If I modify the compute_tile_dm_vcu108.sv to have NUM_CORES parameter equal to 2, the result still remains the same with only one core instantiated. Am I missing something here?

imphil, 2019-01-23:
How do you check that only one core is instantiated? Have a look at the synthesis logs produced by Vivado; it tells you the values assigned to all parameters.

SRQ91, 2019-01-23:
I am checking this by looking at the Hierarchy of sources in Vivado. Under u_compute_tile, I see only one instantiated u_core. I have used this method as for a 2x2 system I used to see multiple u_compute_tile modules instantiated in the sources list.

In synthesis I see that NUM_CORES is bound to 2:
```
INFO: [Synth 8-638] synthesizing module 'compute_tile_dm_vcu108' [/home/saq/build/optimsoc_examples_compute_tile_vcu108_0/src/optimsoc_examples_compute_tile_vcu108_0/rtl/verilog/compute_tile_dm_vcu108.sv:29]
	Parameter NUM_CORES bound to: 2 - type: integer 
	Parameter LMEM_SIZE bound to: 134217728 - type: integer 
```
Is my method of determining the number of instantiated cores incorrect?

imphil, 2019-01-23:
> Is my method of determining the number of instantiated cores incorrect?

no. The source file is included only once. You need to look at the hierarchy after synthesis.

imphil, 2019-01-23:
So this is resolved then, great!

closed 2019-01-23 by imphil