optimsoc/optimsoc #163: [ERROR] libosd: glip: Given speed xx did not work with device. Try autodetect..

SRQ91, 2019-01-28:
I synthesized compute_tile with the following:

`fusesoc --cores-root $HOME/xx/optimsoc/examples/fpga/vcu108/compute_tile build optimsoc:examples:compute_tile_vcu108 --UART0_SOURCE=onboard --HOST_IF=uart`

It works fine and implements the design. For connecting the board I use:

`osd-target-run -e hello.elf -b uart -o device=/dev/ttyUSB1,speed=3000000`

This returns:

```
[ERROR] libosd: glip: Given speed 3000000 did not work with device. Try autodetect..
[ERROR] libosd: glip: Could not autodetect baud rate.
[ERROR] libosd: glip: Cannot open backend (rv=-1)
[ERROR] libosd: Unable to open connection to device (-1)
[FATAL] osd-target-run: Unable to connect to host controller and to device.
```

I have also checked with other speeds just in case. ttyUSB0 is showing me the configuration menu for the FPGA using putty. Am I missing some step which might be causing this issue?

imphil, 2019-01-28:
The onboard UART of the VCU108 doesn't support 3 MBit/s IIRC. The example at https://github.com/optimsoc/optimsoc/tree/master/examples/fpga/vcu108/compute_tile#step-5-run-software uses 921600 baud and should work.

imphil, 2019-01-28:
I just saw that https://github.com/optimsoc/optimsoc/tree/master/examples/fpga/vcu108/compute_tile#onboard-uart gives different instructions. Please double-check with the source code. Could you then open a PR to fix this README? Thanks!

SRQ91, 2019-01-28:
Sure I can open the PR. About the issue, as I mentioned in the original post, I also tried 115200, and 921600 but these don't work too. Another interesting this I just checked is that despite /dev/ttyUSB0 being accessible using putty, the GLIP tool Loopback Measure cannot connect to this port.
`glip_loopback_measure -b uart -ospeed=115200,device=/dev/ttyUSB0`
gives me:
```
glip: gb_uart_open: Given speed 115200 did not work with device. Try autodetect..
glip: gb_uart_open: Could not autodetect baud rate.
glip: glip_open: Cannot open backend (rv=-1)
ERROR: Cannot open backend
```

imphil, 2019-01-28:
Please check the HDL source code with the parameter that's set there and use that speed. Reload the bitstream after you've used a wrong baud rate to ensure that the device is fully reset.

SRQ91, 2019-01-28:
The source code has:

`parameter UART0_BAUD = (UART0_SOURCE == "pmod" ? 3000000 : 921600);`

The 921600 should work but even after reloading I am getting the same issue.

SRQ91, 2019-01-29:
After synthesizing the code for pmod instead of onboard uart, I see only ttyUSB0 connection.

```
$ osd-target-run -e hello.elf -b uart -o device=/dev/ttyUSB0,speed=3000000 -vvv
[INFO]  osd-target-run: iniparser: cannot open /etc/osd/osd.conf
[INFO]  osd-target-run: Unable to parse configuration file /etc/osd/osd.conf
[INFO]  osd-target-run: Stopping all CPUs in the system
[INFO]  osd-target-run: Setting up tracing
[INFO]  osd-target-run: Setting up terminal
[INFO]  osd-target-run: Loading memories ...
[INFO]  osd-target-run: Loading memory at DI address 1 with ELF file hello.elf (not verifying write)
[INFO]  libosd: Load program header 0
[INFO]  libosd: Load program header 1
[INFO]  osd-target-run: Starting all CPUs
[ERROR] libosd: Device read thread did not shut down itself. This typically indicates a bug in the code using osd_gateway.
```

imphil, 2019-01-29:
A couple things to try:
- Of course the PMOD UART is one option, and highly recommended (the onboard UART isn't overly reliable)
- You did log out and log in again after the group change? Changing the device permissions to 0666 might also be worth a try.
- Are you using any hub (or even worse, any USB 3 hubs) in between your board and your PC? Try to directly connect to the PC. Also try different USB cables. We've seen a surprising number of errors due to faulty cables and especially due to faulty hubs.
- If none of that helps, you'll need to resynthesize with ILA cores and observe the UART signals, and possibly other signals, within the design.

SRQ91, 2019-01-30:
For the things that you suggested:

1. I switched to PMOD and as my last post shows, it connects fine but comes to a halt with the error:
`[ERROR] libosd: Device read thread did not shut down itself. This typically indicates a bug in the code using osd_gateway.`

2. I was already in the dialout group, I re-added myself just to make sure of it.

3. The FPGA is connected directly to our server without any hub or other devices in the middle. 

4. I think the UART issue is solved and there is some other issue. As shown below loading memories etc. was done successfully. 

```
$ osd-target-run -e hello.elf -b uart -o device=/dev/ttyUSB0,speed=3000000 -vvv
[INFO]  osd-target-run: iniparser: cannot open /etc/osd/osd.conf
[INFO]  osd-target-run: Unable to parse configuration file /etc/osd/osd.conf
[INFO]  osd-target-run: Stopping all CPUs in the system
[INFO]  osd-target-run: Setting up tracing
[INFO]  osd-target-run: Setting up terminal
[INFO]  osd-target-run: Loading memories ...
[INFO]  osd-target-run: Loading memory at DI address 1 with ELF file hello.elf (not verifying write)
[INFO]  libosd: Load program header 0
[INFO]  libosd: Load program header 1
[INFO]  osd-target-run: Starting all CPUs
[ERROR] libosd: Device read thread did not shut down itself. This typically indicates a bug in the code using osd_gateway.
```

If you agree that this is no longer a UART issue, I will close this issue. (Although the onboard option is still not working.)

imphil, 2019-01-30:
> [ERROR] libosd: Device read thread did not shut down itself. This typically indicates a bug in the code using osd_gateway.

You can ignore that error, it's only telling you that a clean shutdown wasn't possible -- but everything else worked and the shutdown is forced. Good you sorted this out, it confirms our suspicion that the onboard UART of the VCU108 board isn't any good at all.

closed 2019-01-30 by imphil