optimsoc/optimsoc #167: Avoid deadlocks caused by packets exceeding the MP buffer size

koenenwmn, 2019-02-19:
The maximum packet size for the NoC is currently limited by the size of the `noc_buffer` in the `mpbuffer_endpoint`. Larger packets cannot be sent since only full packets are forwarded by the FIFO. However, larger packets are accepted by the `mpbuffer_endpoint` which leads to a deadlock state. Furthermore, the software has no way of knowing how large the buffers are.  
I would suggest to:
1. make the size of the buffer accessible via the baremetal library (similar to how the number of endpoints can be read), and
2. adjust the `mpbuffer_endpoint` so that it discards packets that exceed its buffer size.

imphil, 2019-02-19:
1. It indeed seems to be the case that the maximum packet size isn't accessible. It should be easy to add this parameter to `networkadapter_conf.sv`, which makes it accessible through a memory-mapped register read from the NA.

2. Dropping excessive buffers seems to be a halfway sane thing to do; a real solution would probably involve an error flag and possibly an error interrupt or some other way to get the user's attention. Both deadlocks and silently dropped packets are rather annoying when debugging software, since it often takes a couple trips through the hardware source code to figure out the root cause of such issues.

koenenwmn, 2019-02-20:
I was actually thinking of reading this parameter from `mpbuffer` itself since the number of endpoints is also read from there and both are parameters of the module anyway.

As for the dropping you are right of course. I'm just not sure how to implement such an error flag. What happens if we set the error flag of the bus in such a case? Since sending data is simply a memory-mapped register write and there are no checks.  
We could also explicitly check if the write was successful, or handle it purely in software by reading the buffer size when initializing and then immediately returning an error when a larger message is to be sent.

imphil, 2019-02-20:
> I was actually thinking of reading this parameter from `mpbuffer` itself since the number of endpoints is also read from there and both are parameters of the module anyway.

I'm wondering if the maximum packet size is a NoC parameter/limitation or an limitation of the MP buffer. By grepping for the use of fullpacket = 1 in our source code the answer should become obvious. :-)

> What happens if we set the error flag of the bus in such a case?
In theory you should get a bus exception on the CPU which can be handled there, but I think the system just hangs at the moment.

> Since sending data is simply a memory-mapped register write and there are no checks.
It should be trivial to introduce a control/status register in the mpbuffer with an error bit in it which is always read after a send operation to see if it has failed. (That's a pretty common and sane design in fact.)

koenenwmn, 2019-02-22:
It's a limitation by the MP buffer. Checked the code and also did a quick simulation where I increased the buffer size to 64 and sent packets of that size in the `hello_mpsimple` example.

Alright, I will go for the status register approach then.

wallento, 2019-02-26:
This parameter should be read by the software.