optimsoc/optimsoc #168: Send NoC messages to cores other than core 0

koenenwmn, 2019-02-19:
Currently, arriving NoC packets can only be accepted by core 0 of a tile since no other core receives an interrupt. I was thinking it might be an idea to also allow messages to be accepted by other cores.

This could be achieved by setting the core ID of the receiving core in the endpoint when enabling it. The endpoint would then raise an interrupt at that core when receiving a packet and each core only reads packets from endpoints it has enabled.  
Ideally, there should be some kind of safety mechanism to ensure that only the set core can read from that endpoint.

What do you think @imphil, @wallento?

imphil, 2019-02-19:
The current design isn't uncommon: it requires core 0 to handle all incoming messages, somehow unpack them and forward them to the recipient task -- the mapping between task and core is something only the operating system (running on core 0) knows in most cases.

Of course, depending on your system design the sender might already know the core the recipient runs on (especially if no operating system with a "task" concept is used). In the end, it really depends on your use case. Is this something you currently have a need for, or is this just "thinking ahead"? @wallento might have done work in this regard in his thesis as well.

koenenwmn, 2019-02-20:
We will probably have a use case for a fail-operational scenario where a spare core can take over in case core 0 fails (for whatever reason). I thought it might be useful in other system designs as well. But I think in this case we will just make these adjustments in our project for now.

imphil, 2019-02-20:
If you want to have a fail-operational scenario with core 0 failing you need to route all interrupts to another core, which needs to be done in hardware. And then you most likely inform the "new core 0" about that fact by raising an interrupt to it and through that calling a software routine which handles all further migration tasks.

closed 2019-02-22 by koenenwmn