optimsoc/optimsoc #169: [ERROR] osd-target-run: Unable to load memory at DI address 1 (-13)

SRQ91, 2019-02-19:
I have compiled the standard "hello" example in baremetal-apps and it runs as expected when simulated. However, I get blank system trace files when I load the generated elf on the VCU108. I follow the following process:

`/baremetal-apps/hello$ osd-target-run -e hello.elf -b uart -o device=/dev/ttyUSB0,speed=3000000 --coretrace --systrace --verify  -vvv`

```
[INFO]  osd-target-run: iniparser: cannot open /etc/osd/osd.conf
[INFO]  osd-target-run: Unable to parse configuration file /etc/osd/osd.conf
[INFO]  osd-target-run: Stopping all CPUs in the system
[INFO]  osd-target-run: Setting up tracing
[INFO]  osd-target-run: Writing system trace event output to file systrace.0002.log
[INFO]  osd-target-run: Writing system trace print output to file systrace.print.0002.log
[INFO]  osd-target-run: Writing core trace to file coretrace.0003.log
[INFO]  osd-target-run: Writing system trace event output to file systrace.0004.log
[INFO]  osd-target-run: Writing system trace print output to file systrace.print.0004.log
[INFO]  osd-target-run: Writing core trace to file coretrace.0005.log
[INFO]  osd-target-run: Writing system trace event output to file systrace.0006.log
[INFO]  osd-target-run: Writing system trace print output to file systrace.print.0006.log
[INFO]  osd-target-run: Writing core trace to file coretrace.0007.log
[INFO]  osd-target-run: Writing system trace event output to file systrace.0008.log
[INFO]  osd-target-run: Writing system trace print output to file systrace.print.0008.log
[INFO]  osd-target-run: Writing core trace to file coretrace.0009.log
[INFO]  osd-target-run: Setting up terminal
[INFO]  osd-target-run: Loading memories ...
[INFO]  osd-target-run: Loading memory at DI address 1 with ELF file hello.elf (verifying write through readback)
[INFO]  libosd: Load program header 0
[INFO]  libosd: Load program header 1
[INFO]  libosd: Verify program header 0
[ERROR] libosd: Memory mismatch at byte 0x57f. Expected 10, read 00
[ERROR] osd-target-run: Unable to load memory at DI address 1 (-13)
[INFO]  osd-target-run: Starting all CPUs
[INFO]  osd-target-run: System is now running. Press CTRL-C to end tracing.
^C[INFO]  osd-target-run: Shutdown signal received, cleaning up.
[INFO]  libosd: Systracelogger statistics: 0 overflowed packets, 0 trace events, 0 sysprint events
[INFO]  libosd: Systracelogger statistics: 0 overflowed packets, 0 trace events, 0 sysprint events
[INFO]  libosd: Systracelogger statistics: 0 overflowed packets, 0 trace events, 0 sysprint events
[INFO]  libosd: Systracelogger statistics: 0 overflowed packets, 0 trace events, 0 sysprint events
[ERROR] libosd: Device read thread did not shut down itself. This typically indicates a bug in the code using osd_gateway.
```

I think the error might be due to the vmem file not being loaded as already evident by the output. The core traces show relevant data. I can post the traces if required.

imphil, 2019-02-19:
`[ERROR] libosd: Memory mismatch at byte 0x57f. Expected 10, read 00`
indicates the memory write didn't go well. Typically this is caused by the off-chip connection. If this problem happens all the time try to reduce the off-chip speed to less than 3 MBaud (both in software and when calling osd-target-run), or use the USB3 off-chip interface, which we recommend with this board.

SRQ91, 2019-02-20:
I am using the Digilent pmod device to connect to the FPGA. I resynthesized the system for baudrate of 921600. It did not resolve the issue. Details are as follows:

`/baremetal-apps/hello$ osd-target-run -e hello.elf -b uart -o device=/dev/ttyUSB0,speed=921600 --coretrace --systrace --verify  -vvv`

```
[INFO]  osd-target-run: iniparser: cannot open /etc/osd/osd.conf
[INFO]  osd-target-run: Unable to parse configuration file /etc/osd/osd.conf
[INFO]  osd-target-run: Stopping all CPUs in the system
[INFO]  osd-target-run: Setting up tracing
[INFO]  osd-target-run: Writing system trace event output to file systrace.0002.log
[INFO]  osd-target-run: Writing system trace print output to file systrace.print.0002.log
[INFO]  osd-target-run: Writing core trace to file coretrace.0003.log
[INFO]  osd-target-run: Writing system trace event output to file systrace.0004.log
[INFO]  osd-target-run: Writing system trace print output to file systrace.print.0004.log
[INFO]  osd-target-run: Writing core trace to file coretrace.0005.log
[INFO]  osd-target-run: Writing system trace event output to file systrace.0006.log
[INFO]  osd-target-run: Writing system trace print output to file systrace.print.0006.log
[INFO]  osd-target-run: Writing core trace to file coretrace.0007.log
[INFO]  osd-target-run: Writing system trace event output to file systrace.0008.log
[INFO]  osd-target-run: Writing system trace print output to file systrace.print.0008.log
[INFO]  osd-target-run: Writing core trace to file coretrace.0009.log
[INFO]  osd-target-run: Setting up terminal
[INFO]  osd-target-run: Loading memories ...
[INFO]  osd-target-run: Loading memory at DI address 1 with ELF file hello.elf (verifying write through readback)
[INFO]  libosd: Load program header 0
[INFO]  libosd: Load program header 1
[INFO]  libosd: Verify program header 0
[ERROR] libosd: Memory mismatch at byte 0x46. Expected 00, read 20
[ERROR] osd-target-run: Unable to load memory at DI address 1 (-13)
[INFO]  osd-target-run: Starting all CPUs
[INFO]  osd-target-run: System is now running. Press CTRL-C to end tracing.
^C[INFO]  osd-target-run: Shutdown signal received, cleaning up.
[INFO]  libosd: Systracelogger statistics: 0 overflowed packets, 5 trace events, 0 sysprint events
[INFO]  libosd: Systracelogger statistics: 0 overflowed packets, 3 trace events, 0 sysprint events
[INFO]  libosd: Systracelogger statistics: 0 overflowed packets, 3 trace events, 0 sysprint events
[INFO]  libosd: Systracelogger statistics: 0 overflowed packets, 3 trace events, 0 sysprint events
[ERROR] libosd: Device read thread did not shut down itself. This typically indicates a bug in the code using osd_gateway.
```

When you said to reduce the baud rate in Software, you meant the Verilog code, right?

imphil, 2019-02-20:
> When you said to reduce the baud rate in Software, you meant the Verilog code, right?

yes, that was a typo on my side, sorry for that. I meant "both in hardware and when calling osd-target-run", with hardware being the Verilog code.

Unfortunately, it's a bit hard to predict what's going wrong here -- communication bugs are the most annoying ones. You could try:

- Try different cables, USB ports, etc. It could be just an unreliable cable, switch or anything in between. [Takes a couple of hours to test.]
- Test the off-chip connection itself by running glip's loopback demo and the stress test (it's in the glip source tree): https://www.glip.io/group__backend__uart-examples-vcu108__loopback.html If these tests find anything suspicious then you'd need to further look into them by using on-chip instrumentation (ILAs in Vivado) and see where the problem comes from (could be in the GLIP software code, or in the hardware) [Can take ~ 1 day to test.]
- If you don't find anything in GLIP then things get even more annoying. It could be a communication bug that appears only when used in OpTiMSoC, or it could be somewhere totally else. If that's the case, you first need to find a way to reliably trigger Vivado's ILAs on the point where the memory write verification error happens, and then go back from there to find the root cause. [Can easily take a week to figure out.]

If you need something fast you need to look into workarounds. As you already moved from workaround 1 to workaround 2 (different UART adapter), workaround 3 would be the FX3 USB setup.

imphil, 2019-02-20:
Just to make sure I didn't miss anything: @wallento what you thought might be a GLIP UART bug was in fact no bug?

SRQ91, 2019-02-20:
So here are some initial findings:

1. I changed 5 USB cables, the same issue persists. NOT a cable issue.

2. To test the diligent pmod itself, I shorted the RX and TX pin of the board and sent data via HTTerminal. All data was correctly received. NOT a board issue.

3. GLIP debugging:
a. Stress test:
`/baremetal-apps/hello$ glip_io_stress_test -b uart -ospeed=921600,device=/dev/ttyUSB0`
OUTPUT:
```
Detected interface width: 16-bit.
Running stress test with 10 MB of data, writing 2048 bytes in one block, 
and reading 2048 bytes in one block.
   0 % [                                                  ]   34 KiB TX 
Error reading from GLIP device during write test. Received invalid error message from device: 0x0200
```

b. Loopback Measure:
`/baremetal-apps/hello$ glip_loopback_measure -b uart -ospeed=921600,device=/dev/ttyUSB0`
OUTPUT:
```
Running loopback test with 10485760 bytes of data, writing 256 bytes in one block, 
and reading 2048 bytes in one block.
Data verification failed: expected 0x1, got 0x5 at byte 1. Next byte: 0x6 K
```

imphil, 2019-02-20:
So this looks like a GLIP issue, possibly specific to this device. Is is reproducible, i.e. always happening at the same at the same time? Beyond that you need to unfortunately go into debugging the GLIP UART backend itself. @koenenwmn has some experience with that. (It would be good to first confirm at TUM with another VCU108 board with pmod UART to see if the issue happens there as well.)

SRQ91, 2019-02-20:
At my side the issue is reproducible and loading any elf file with baudrate of 921600 gives this problem consistently at memory location 0x46. The error always remains:
`[ERROR] libosd: Memory mismatch at byte 0x46. Expected 00, read 20`
which is a bit strange. @koenenwmn since you have another VCU108, can you please recreate this scenario. I have synthesized a single tile 4 core system with pmod @ 921600.

imphil, 2019-02-20:
Ignore the optimsoc use case for now until the glip stress and loopback tests fully work. Do you get the same error position of every run of the stress test and (even better) in the loopback test?

SRQ91, 2019-02-20:
Yes. We have another VCU108 that was fortunately free. I loaded the same bitstream on it. The same issue persists. The only difference is the memory address where the data mismatch occurs. The error for this board for any elf is:

```
[ERROR] libosd: Memory mismatch at byte 0x86. Expected 00, read 12
[ERROR] osd-target-run: Unable to load memory at DI address 1 (-13)
```

SRQ91, 2019-02-20:
> Ignore the optimsoc use case for now until the glip stress and loopback tests fully work. Do you get the same error position of every run of the stress test and (even better) in the loopback test?

My error position do you mean the memory address at which the error occurs? If yes, then this address is not the same. For the glip loopback measure the error is detected already at byte 1.

imphil, 2019-02-20:
> By error position do you mean the memory address at which the error occurs?

That's optimsoc. Ignore optimsoc for now and *only* debug GLIP until it works.

Btw, just to ensure that: Do you reflash the FPGA and possibly reset it (using the CPU reset button) between tries to ensure that you have a clean environment?

imphil, 2019-02-20:
And for the GLIP tests you built and flashed the appropriate bitstreams from the GLIP repository?

SRQ91, 2019-02-20:
As far as flashing FPGA is concerned, I did reset it regularly as after the glip commands I had issues running osd-target-run. For GLIP I have pre-generated bitstreams. I will generate them again in case something was changed.

koenenwmn, 2019-02-21:
I tested the glip stress test and the single compute tile example (with 1 core) both using the pmod and a speed of 3MBaud. Both worked fine (had a connection issue with the compute tile example but a rebuild of OpTiMSoC fixed that).  
Did you make sure to rebuild everything after pulling the latest master?

koenenwmn, 2019-02-22:
A little update:  
I managed to run the example on a compute tile with 1,2,3, and 4 cores now. Yesterday I ran into a similar issue you described initially, but with a 3 core system (and I couldn't connect to a 4 core system at all). Updating the mor1kx seems to have fixed that issue, so you should try the current master.

I would still strongly recommend using the USB3 off-chip interface since even with 3MBaud this rather simple example causes overflows as soon as 2 cores are present and doing printfs.

SRQ91, 2019-02-26:
I tried again with the master to run the hello program using the single tile quad-core system. It connects fine but has the same memory error. I will order the Cypress FX3 USB 3.0 kit.

SRQ91, 2019-03-07:
I can confirm that with Cypress FX3 the loading is fine and I see accurate system traces. The issue can be closed for now and revisited later on if pmod has to be made operational for other reasons. Thank you for all the help.

closed 2019-03-07 by SRQ91