optimsoc/optimsoc #174: Xilinx IPs

koenenwmn, 2019-03-08:
The current implementation of the system_2x2_cccc does not work on the VCU108. The reason is that the Xilinx IPs for the `xilinx_axi_register_slice` and the `xilinx_axi_interconnect_4to1` have been designed for the nexys4ddr board and mismatch the required address width (which causes the areas of all 4 tiles to be mapped to the same address in the DRAM). I created new versions for the VCU108 and it works that way.

My question is: wouldn't it make more sense to move the currently used Xilinx IPs for the `xilinx_axi_register_slice` and `xilinx_axi_interconnect_4to1` to the board support in the fusesoc_cores repository (since they are specific to the nexys4ddr example) and also add the corresponding versions for the VCU108 there?

What do you think @imphil @wallento? Just wanted to check with you before I open any PRs.

imphil, 2019-03-08:
I'm not a big fan of the fusesoc_cores repo, it's external but has no other users than us. So I'd go the other way, and fold this repo back into the optimsoc repository. 

The general idea of the board support package was to abstract away board differences under a common interface. The AXI interconnect IP isn't really board specific, but more target and design-specific. I'd therefore place it in the toplevel design directory.

koenenwmn, 2019-03-08:
Ok. For now I would then just move the current IPs to the toplevel of the nexys4ddr example and add new ones for the vcu108 example. Then we can still reintegrate the fusesoc_cores repo into OpTiMSoC at a later point (just want to get the examples running for now).

koenenwmn, 2019-03-08:
Nevermind, since I have to adjust the IPs used in the vcu108 board wrapper I'll give it a shot right away.

closed 2019-03-08 by koenenwmn