optimsoc/optimsoc #178: Modifying the compute tile / Reusing the mor1kx with NA

APDNath, 2019-04-16:
Hello, 

I am just wondering what would be the best approach to reuse the mor1kx CPU and the associated network adapter as individual IPs connected to a NoC router. I have a couple  of questions regarding this: 
1. Can I use just the mor1kx cpu and associated network adapter and get rid of the wb_bus, sram, and memory adapter modules from the compute tile?
2. If so, would I still be able to connect the CPU and NA to a router and make those function?

Any help/suggestion/comment would be much appreciated. 

Thanks!

imphil, 2019-04-26:
The CPU and the network adapter are connected through the bus. If you don't want to use the wishbone bus, you need to provide your own replacement. Furthermore, if you don't want to hook up the SRAM to the bus you need to provide another way for the CPU to fetch its program code and data, e.g. through by extending the load/store unit in the network adapter.

APDNath, 2019-04-26:
Thanks much for the reply. It is really helpful.

closed 2019-04-26 by imphil