optimsoc/optimsoc #180: Query about peripheral integration

APDNath, 2019-07-17:
Hello, 

I am just wondering what are the required modifications to hook up a wishbone compatible peripheral like AES/DSP block to the network adapter of the OptimSoC and make it functional/accessible as a slave to the master i.e. mor1kx?  Is it similar to connecting the in flit / out flit signals of the network adapter to the peripheral like the SRAM through a memory adapter? 
Any suggestion would be appreciated. 

Thanks!

imphil, 2019-08-12:
You only need to add the peripheral to the bus within compute_tile_dm.sv, and you need to change the bus configuration to have room for one more slave (here: https://github.com/optimsoc/optimsoc/blob/master/src/soc/hw/compute_tile_dm/verilog/compute_tile_dm.sv#L414)

APDNath, 2019-08-21:
Thanks much for the reply. I am just wondering what kinda change/wrapper addition is required to the accelerator tile to make it compatible to the NoC protocol and message passing so that there is no requirement for the wishbone bus. Also, are you guys planning to release such an accelerator tile in the near future? I have attached a slide from your presentation for reference.  
[image]

imphil, 2019-08-21:
There's no example of such an accelerator currently available. There is documentation at https://www.optimsoc.org/docs/master/refman/networkonchip.html#transmission-packet-format about the NoC format, you can use that together with the building blocks we have in OpTiMSoC to build your own accelerator.

panchapandava, 2019-08-21:
Thank for the quick reply. It’s a useful piece of information.

closed 2019-08-21 by panchapandava