optimsoc/optimsoc #181: Invalid website digital SSL certificate

lucasandradecb, 2019-10-08:
Hello,
I am working on some projects using your OptimSoc tool. However yesterday, when trying to access the site (https://optimsoc.org/docs/master/refman/porting.html) I received the 404 message from the browser, in addition to the unsafe site information due to problems with the invalid SSL certificate. When checking by the browser it was renewed on 06/10/2019, but it was not validated. With this, so far the site is inaccessible.
Would it be possible to rehabilitate access to the site?

Thank you very much in advance.

wallento, 2019-10-08:
Hi, sorry, it is temporarily down. It will return asap.

wallento, 2019-10-15:
Sorry, forgot to ping you. The issue was fixed, thanks again.

closed 2019-10-15 by wallento