optimsoc/optimsoc #184: Doc build failure, possibly due to Python package updates

imphil, 2019-11-25:
See https://travis-ci.org/optimsoc/optimsoc/builds/616690987?utm_medium=notification&utm_source=email

We see this in unchanged code on Travis and in internal CI, so it's an external dependency that changed.

imphil, 2019-11-25:
It started to fail 2 days ago.

imphil, 2019-11-25:
```
Collecting cairocffi
  Downloading https://files.pythonhosted.org/packages/f7/99/b3a2c6393563ccbe081ffcceb359ec27a6227792c5169604c1bd8128031a/cairocffi-1.1.0.tar.gz (68kB)
    ERROR: Command errored out with exit status 1:
     command: /src/objdir/doc/.venv/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-e4es9hzy/cairocffi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-e4es9hzy/cairocffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-e4es9hzy/cairocffi/pip-egg-info
         cwd: /tmp/pip-install-e4es9hzy/cairocffi/
    Complete output (46 lines):
    WARNING: The wheel package is not available.
    WARNING: The wheel package is not available.
    WARNING: The wheel package is not available.
      ERROR: Command errored out with exit status 1:
       command: /src/objdir/doc/.venv/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-wheel-0d6tl79m/pycparser/setup.py'"'"'; __file__='"'"'/tmp/pip-wheel-0d6tl79m/pycparser/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-6lyln84u
           cwd: /tmp/pip-wheel-0d6tl79m/pycparser/
      Complete output (6 lines):
      usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
         or: setup.py --help [cmd1 cmd2 ...]
         or: setup.py --help-commands
         or: setup.py cmd --help
    
      error: invalid command 'bdist_wheel'
      ----------------------------------------
      ERROR: Failed building wheel for pycparser
    ERROR: Failed to build one or more wheels
    Traceback (most recent call last):
      File "/src/objdir/doc/.venv/lib/python3.5/site-packages/setuptools/installer.py", line 119, in fetch_build_egg
        subprocess.check_call(cmd)
      File "/usr/lib/python3.5/subprocess.py", line 581, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/src/objdir/doc/.venv/bin/python3', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpzqpsx6b8', '--quiet', 'pycparser']' returned non-zero exit status 1
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-e4es9hzy/cairocffi/setup.py", line 13, in <module>
        'cairocffi/ffi_build.py:ffi_pixbuf']
      File "/src/objdir/doc/.venv/lib/python3.5/site-packages/setuptools/__init__.py", line 144, in setup
        _install_setup_requires(attrs)
      File "/src/objdir/doc/.venv/lib/python3.5/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/src/objdir/doc/.venv/lib/python3.5/site-packages/setuptools/dist.py", line 721, in fetch_build_eggs
        replace_conflicting=True,
      File "/src/objdir/doc/.venv/lib/python3.5/site-packages/pkg_resources/__init__.py", line 782, in resolve
        replace_conflicting=replace_conflicting
      File "/src/objdir/doc/.venv/lib/python3.5/site-packages/pkg_resources/__init__.py", line 1065, in best_match
        return self.obtain(req, installer)
      File "/src/objdir/doc/.venv/lib/python3.5/site-packages/pkg_resources/__init__.py", line 1077, in obtain
        return installer(requirement)
      File "/src/objdir/doc/.venv/lib/python3.5/site-packages/setuptools/dist.py", line 777, in fetch_build_egg
        return fetch_build_egg(self, req)
      File "/src/objdir/doc/.venv/lib/python3.5/site-packages/setuptools/installer.py", line 121, in fetch_build_egg
        raise DistutilsError(str(e))
    distutils.errors.DistutilsError: Command '['/src/objdir/doc/.venv/bin/python3', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmpzqpsx6b8', '--quiet', 'pycparser']' returned non-zero exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

imphil, 2019-11-25:
OK, so this falls together with setuptools 42.0.0 being released, with 42.0.1 being released a couple hours ago with a wheel-related fix. Let's see if that unbreaks us.

imphil, 2019-11-25:
OK, setuptools 42.0.1 doesn't solve the problem, https://travis-ci.org/optimsoc/optimsoc/builds/616755658?utm_medium=notification&utm_source=email still fails.

benoit-pierre, 2019-11-25:
See https://github.com/pypa/setuptools/issues/1919.

imphil, 2019-11-26:
@benoit-pierre thanks for the pointer. What I don't understand: if wheel is required, why isn't it installed just like any other dependency?

benoit-pierre, 2019-11-26:
It's not always required, you don't need it to install already built wheels, or to build wheels for packages using PEP 517/518. Trying to automatically install it could lead to other issues.

PR #185 (2019-12-02): Work around cairocffi not being installable by setuptools
setuptools 42.0.0 and 42.0.1 isn't able to install cairocffi, work
around that by disallowing any 42.0.* version for now.

Fixes #184
commit 7f506eda (2019-12-02): Work around cairocffi not being installable by setuptools

setuptools 42.0.0 and 42.0.1 isn't able to install cairocffi, work
around that by disallowing any 42.0.* version for now.

Fixes #184
closed 2019-12-02 by imphil