optimsoc/optimsoc #186: osd-target-run: error while loading shared libraries: libczmq.so.3: cannot open shared object file: No such file or directory

IMVector, 2019-12-21:
when I try to run Linux on OpTiMSoC, I found this error and I can't resove it，Can anyone help me out？

imphil, 2019-12-21:
You need to install libczmq, which is part of zero mq. Which distribution are you using? How did you obtain optimsoc?

IMVector, 2019-12-21:
> You need to install libczmq, which is part of zero mq. Which distribution are you using? How did you obtain optimsoc?

My system is Ubuntu 18.04, I got the optimsoc from https://www.optimsoc.org/docs/master/user_guide/tutorials.html#run-linux-on-optimsoc

I solved the problem, but here comes a new problem.
when I try to run linux on my board 
I got this 
------------------------------------------------------------------------------------
[INFO]  osd-target-run: iniparser: cannot open /etc/osd/osd.conf
[INFO]  osd-target-run: Unable to parse configuration file /etc/osd/osd.conf
[INFO]  osd-target-run: Stopping all CPUs in the system
[INFO]  osd-target-run: Setting up tracing
[INFO]  osd-target-run: Writing system trace event output to file systrace.0002.log
[INFO]  osd-target-run: Writing system trace print output to file systrace.print.0002.log
[INFO]  osd-target-run: Setting up terminal
[INFO]  osd-target-run: Loading memories ...
[INFO]  osd-target-run: Loading memory at DI address 1 with ELF file /home/vector/buildroot/output/images/vmlinux (not verifying write)
[INFO]  libosd: Load program header 0
[INFO]  libosd: Load program header 1
[INFO]  libosd: Load program header 2
[INFO]  osd-target-run: Starting all CPUs
[INFO]  osd-target-run: System is now running. Press CTRL-C to end tracing.

------------------------------------------------------------------------------------
and I can't boot the linux

imphil, 2019-12-21:
Your log doesn't show any error, it seems to work as expected. Did you try the hello world example before trying Linux? How do you know that booting linux didn't work? Did you connect via UART? What are the log files (systrace*.log) showing?

IMVector, 2019-12-21:
systrace*.log shows nothing and I didn't get the information of  DEM-UART pseudo-terminal available at /dev/pts/"number". I am a new learner and don't know what is UART, I run the hello world example successfully, but when I try to run Linux , I didn't get the result as the tutorials

------------------------------------------------------
Watch the output of this command. If all goes well the output should contain a line similar to libosd: DEM-UART pseudo-terminal available at /dev/pts/19. Keep note of the device file (starting with /dev/pts/), you’ll need this path to connect to the Linux console on the OpTiMSoC system.
------------------------------------------------------

IMVector, 2019-12-21:
> Your log doesn't show any error, it seems to work as expected. Did you try the hello world example before trying Linux? How do you know that booting linux didn't work? Did you connect via UART? What are the log files (systrace*.log) showing?

systrace*.log shows nothing and I didn't get the information of DEM-UART pseudo-terminal available at /dev/pts/"number". I am a new learner and don't know what is UART, I run the hello world example successfully, but when I try to run Linux , I didn't get the result as the tutorials
-------------------------------------------------------------------------------------------------------
Watch the output of this command. If all goes well the output should contain a line similar to libosd: DEM-UART pseudo-terminal available at /dev/pts/19. Keep note of the device file (starting with /dev/pts/), you’ll need this path to connect to the Linux console on the OpTiMSoC system.

imphil, 2020-01-06:
UART is a serial console. Please google it.

To access a serial console, you need to use an appropriate program, like screen. There are many more options, please google for it.

IMVector, 2020-01-07:
> UART is a serial console. Please google it.
> 
> To access a serial console, you need to use an appropriate program, like screen. There are many more options, please google for it.

Thank you very much ， I try it last time，but the result shown as above.