optimsoc/optimsoc #187: Testbench for OptimSoC

panchapandava, 2020-02-11:
Hello, 

I am just wondering if it is possible to get some pointers on the system -level testbench (preferably Verilog/SystemVerilog) available for OptimSoC. I couldn't find any in the repository. 
Also, I am not sure how to use "optimsoc/test/systemtest/test_tutorial.data" to exercise different test cases. Any help would be appreciated. 

Thanks!

imphil, 2020-02-11:
The system tests are python-based tests running on a complete Verilator and FPGA system. You can find a blog post about the complete setup here: https://optimsoc.org/blog/2018/12/18/ci.html

Effectively, you run those tests through pytest, as shown here: https://github.com/optimsoc/optimsoc/blob/master/Dockerfile#L32

panchapandava, 2020-02-11:
Thanks much for the quick reply. I will run the python-based tests. Also, can you please comment on the feasibility of writing Verilog/SystemVerilog-based directed testbench for OptimSoC? As in how complex the task would be?

imphil, 2020-02-11:
It really depends on what you want to test. To reach 100% coverage on OpTiMSoC, you're looking for a project of a year at least.

panchapandava, 2020-02-11:
Thanks again for the suggestion. I guess I will try out random and constraint random testbenches to reduce the complexity of the task.

closed 2020-02-11 by panchapandava