optimsoc/optimsoc #189: optimsoc & synopsys vcs and DC; packet format change to 10 bit src/dst

hchenji, 2020-11-03:
Hi all, great project!

I'm trying to use optimsoc in the synopsys toolchain and I'm seeing errors. In VCS I got the errors to go away by replacing the always_ff block in sram to an always block. Then, the clog2 function within an array access was extracted to a separate variable. 

However, the resulting vcs binary fails to correctly simulate the NoC while the same code works well in verilator. What happens is the first few packets (with class set to 7) do not make it to the destination at all in a NoC, even creating duplicate packets. I have added some $display lines to help trace the packets. The hello world programs run fine though. 

Wanted to see if anyone has had any luck with using optimsoc on Synopsys tools. 

The code with vcs adjustments that get it to compile but not simulate correctly in VCS is here: https://github.com/hchenji/optimsoc/commit/7c64becea0f2d03b8fd1b1c7548a4b6807e6114f 

BTW, I took a shot at changing the packet format to have 10 bit IDs for src and dest. It works for a 10x10 NoC in verilator. The code for new packet format is here: https://github.com/hchenji/optimsoc/commit/2e3efe7ac9571614f91bb3a0ce3805d531521526

deadspace99, 2020-11-08:
Hello, I too downloaded this to run through VCS but it looks like you were able to get it through with some modifications.  I had to build a .f filelist  from looking through the hierarchy and the instantiations but I'm running into some errors related to the functions used.  Did you also run into these errors?  Is there a .f that you can share that I can cross-check to make sure nothing was missed?

Error-[XMRENRM] Cross-module reference resolution error
Cross-module reference resolution error: wrong target type.
The target of the reference is expected to be net, register or memory.
Source info: CONFIG.CORES_PER_TILE
Cross-module reference resolution error: wrong target type.
The target of the reference is expected to be net, register or memory.
Source info: CONFIG.CORES_PER_TILE

Error-[CECFC] Cannot evaluate constant function call
"optimsoc_functions::index2string(((0 * CONFIG.CORES_PER_TILE) + 0))"
The constant function call could not be evaluated during elaboration.
Source info: optimsoc_functions::index2string(((0 * CONFIG.CORES_PER_TILE) +
0))

hchenji, 2020-11-09:
I actually use fusesoc to build the vcs version. First, use this version of my forked repo, which has modified .core files to include vcs as a tool: https://github.com/hchenji/optimsoc/commit/7c64becea0f2d03b8fd1b1c7548a4b6807e6114f 

Next, build optimsoc as usual using build.py so that the objdir folder and tree is created. Then, use this script in an empty directory (it could be anywhere) to get fusesoc to use vcs to build, where $OPTIM_ROOT is the location of the optimsoc src tree. It could be possible that build.py also is able to build using vcs given the changes to .core files, but I don't remember. Finally, the .vmem file (from baremetal_apps) needs to be in the same folder as the generated simv binary - name it sram.vmem. 

```
#!/bin/bash

export OPTIMSOC="$OPTIM_ROOT/objdir/dist"
export OPTIMSOC_RTL="$OPTIM_ROOT/objdir/dist/soc/hw"
export LISNOC="$OPTIM_ROOT/objdir/dist/external/lisnoc"
export LISNOC_RTL="$OPTIM_ROOT/objdir/dist/external/lisnoc/rtl"
export FUSESOC_CORES="$OPTIM_ROOT/objdir/dist/soc/hw:$OPTIM_ROOT/objdir/dist/external/lisnoc:$OPTIM_ROOT/objdir/dist/external/opensocdebug/hardware:$OPTIM_ROOT/objdir/dist/external/mor1kx:$OPTIM_ROOT/objdir/dist/external/glip"

fusesoc --verbose --monochrome --cores-root $OPTIM_ROOT/examples/sim/compute_tile sim --build-only optimsoc:examples:compute_tile_sim --NUM_CORES 1
```

deadspace99, 2020-11-20:
Thanks for the guidance.  I was able to get the VCS compile but I don't have the .vmem file.  Could you explain what your reference to the baremetal_apps is and how to generate this?

hchenji, 2020-11-24:
in the optimsoc tutorial, the baremetal-apps repo has code to generate binaries which will run on the cpu cores in each tile. Filename ends in vmem. Just take that and use it for vcs.

imphil, 2020-11-25:
@hchenji We have used VCS for parts of the development, but probably never ran a w hoel

imphil, 2020-11-25:
@koenenwmn have you been using VCS more recently? If you have relevant changes, could you push them as well?

koenenwmn, 2020-12-02:
Only for the cocotb CI tests, never for a full system. And recently I only worked with Verilator simulations and Vivado synthesis.

hchenji, 2020-12-10:
Running hello world.vmem with vcs. Using trace, the CPU is stuck in this loop of instructions and never terminates. Any guess why this happens with vcs but not verilator?

```
[18117218, 0]   5, 0x0000dc0c
[18117222, 0]   1, 0x0000d7cc
[18117244, 0]   6, 0x0000d7c4
[18117248, 0]   1, 0x0000dbec
[18117270, 0]   5, 0x0000dc0c
[18117274, 0]   1, 0x0000d7cc
[18117296, 0]   6, 0x0000d7c4
[18117300, 0]   1, 0x0000dbec
[18117322, 0]   5, 0x0000dc0c
[18117326, 0]   1, 0x0000d7cc
[18117348, 0]   6, 0x0000d7c4
[18117352, 0]   1, 0x0000dbec
[18117374, 0]   5, 0x0000dc0c
[18117378, 0]   1, 0x0000d7cc
[18117400, 0]   6, 0x0000d7c4
[18117404, 0]   1, 0x0000dbec
[18117426, 0]   5, 0x0000dc0c
[18117430, 0]   1, 0x0000d7cc
[18117452, 0]   6, 0x0000d7c4
[18117456, 0]   1, 0x0000dbec
```

hchenji, 2021-01-16:
I was able to get compute_tile_dm to synthesize in synopsys DC. The main change was to disable per-tile SRAM by setting ENABLE_DRAM to 0 in the config since it slowed things down. Next, in mor1kx, the true dual port single clock ram was throwing errors. I had to force a single clk variable, and put both ports into a single always block. the _dm module needs to have a derived CONFIG since the tb isn't going to be there. 

I use mflowgen to compile it. Instructions: clone my fork of mflowgen/mflowgen. In designs/optimsoc/rtl/outputs, create a src symlink to optimsoc source tree. then in mflowgen root, follow the mflowgen tutorial. 

Changes to optimsoc are in my fork of optimsoc: hchenji/optimsoc